OpenEVSE/openevse_esp32_firmware #376: claim engine should deep merge properties

KipK, 2022-06-07:
When making a new claim over an existing one, it doesn't keep previous properties and overwrite them.
It should deep merge json data to avoid sending again the whole json. This will be a pain to integrate mqtt feeds to smart home solutions if we need to resend all claim data for each separated command.

jeremypoulter, 2022-06-11:
Yeah, that is probably ok, the only problem would be how to independently clear a property, eg, set the max current, set the charge current, clear the max current...

KipK, 2022-06-11:
I see EvseProperties::deserialize(JsonObject &obj) doing exactly that isn't it? Shouldn't be a problem.

Merge should be done here and we're good:

```
bool EvseManager::Claim::claim(EvseClient client, int priority, EvseProperties &target)
{
  if(_client != client ||
     _priority != priority ||
     _properties != target)
  {
    _client = client;
    _priority = priority;
    _properties = target;
    return true;
  }

```

KipK, 2022-06-16:
seems we need an undefined state for some evsemanager methods like "isAutoRelease()"
 true/false is not sufficient to know if the current/previous claim has autorelease settings or not.

jeremypoulter, 2022-06-17:
I think it is less of a big deal for HTTP. You will be using it a slightly different way. Under HTTP you would also be sending a client ID (via the path). As any single client will probably only be changing the same things it is less of an issue as the EVSE manager then does the merging/prioritisation.

I guess we could expand MQTT to also have the client ID in the topic name, but then technically it should event all updates to claims from other parts of the system. Not sure how useful that would be.

I do see your point about it not being quite as elegant, maybe some of the text processing could be moved to EvseProperties? Although that being said I think EvseProperties::deserialize does what you want, so just keeping the global and repeatedly calling deserialize on it will merge, but you still have the clearing problem then ...

Also I did mention elsewhere that maybe the full override shouldn't be exposed over MQTT, just a simple `active/disable/toggle/clear` on the `manual_override/set` topic.

KipK, 2022-06-20:
Not sure we need to treat all claims with client id on mqtt side. But can be achieved easily if needed, let's keep it like this for now and see if some users really needs full mqtt control later.

I've removed the settable properties with independant topics. Now merge works ok since this commit : https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/379/commits/214ceec6cc5c4c89dec85ba41965d443d3c53841
I had to modify a bit the evseProperties::deserialize to add a "clear" parameter. 

Let me know what you think of this approach to clear properties independantly

KipK, 2022-06-26:
I've closed it as it's solved in new mqtt branch

closed 2022-06-26 by KipK

KipK, 2022-10-11:
> Maybe we limit this to MQTT? I think that is the main issue right? How
> about /claim/set takes in a complete JSON object and over-writes all
> settings, then we also have /claim/set/charge_current would set the
> specific value preserving any others? Should be fairly easy to implement
> with a global of the MQTT claim in the MQTT module

I think it would have sense on http API too that's why I think it should be
better to be implemented EvseManager side.

PR #379 (2022-07-29): Mqtt reworked API version
Fixes #358 
Fixes #376 
Fixes #373 

Following previous discussion here https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/369 
Based on espidf4 branch.
 
Edit: all functional