OpenEVSE/openevse_esp32_firmware #385: Multi-access point: weakest signal chosen

progman32, 2022-07-04:
I have a commercial WiFi setup with multiple access points creating a single network (happens to be Aruba Network gear). All my devices correctly connect to the strongest access point, except OpenEVSE - it seems to default to the _weakest_ signal. If I temporarily power down all access points other than the one I want OpenEVSE to use, it does associate with the correct access point.... but then goes right back to the weakest one on reboot. Very strange. Does it somehow remember the MAC of the first access point it saw?

The issue persists even if I associate with a completely different network, then reconnect to the original multi-access-point network.

Aside, as a workaround... is there any way to force association with a particular access point, such as by MAC?

davethetallguy, 2022-07-04:
I had a similar problem, though in my case with Ubiquity as the network vendor. 

Ubiquity allowed me to limit which AP OpenEVSE would associate with. Aruba might have the same?

Ultimately I created a unique SSID that only the single AP near the OpenEVSE broadcasts. 

Obviously both are workarounds, but it's been stable.

progman32, 2022-07-04:
Good workaround idea. My Aruba doesn't seem to have such an option, sadly.

chris1howell, 2022-07-04:
Looks like default behavior is to connect to the AP with the lowest
channel. Looks like there is a way to scan signal strength also.

https://www.reddit.com/r/esp32/comments/nphj1c/how_to_connect_to_wifi_ap_has_stronger_signal/

jeremypoulter, 2022-07-05:
Will look in to this, how are you checking the AP that is connected to, from the AP itself?

progman32, 2022-07-05:
> how are you checking the AP that is connected to

From my AP's management interface.

mmiller7, 2022-07-17:
FWIW I found with mine some time ago it was connecting to the lowest channel number in range of multiple BSSIDs that it saw (regardless of signal strength).

My workaround was set up 2 networks - one with the BSSID, another the same SSID and no BSSID at lower priority.  That way it roams if the expected AP goes down.  Then on the "server side" I have HomeAssistant issue a reboot to ESPHome if it notices an access point go from "down" to "up" on the network (tested by pinging its management interface).

Really awful workaround but until they fix the bug for finding the strongest channel dynamically I couldn't find a better way.

jeremypoulter, 2023-02-25:
Thanks to @KipK for finding the solution to this,  will be part of #533.

For the record by default the Arduino library is setup to connect to the first matching SSID it finds, rather than scanning all available APs...

closed 2023-03-31 by glynhudson