OpenEVSE/openevse_esp32_firmware #386: The delay timer does not work

DmitryAVR, 2022-07-04:
Hello. I built OpenEVSE without a display. And the delay timer does not work for me (in the screenshot). 

[image]

When the timer is set, EVSE goes into standby mode. But charging does not start at the beginning of time. Is it due to the display not connected, or do I have another error?

jeremypoulter, 2022-07-05:
Are you using any other 'smart' features, like PV divert?

jeremypoulter, 2022-07-06:
Interesting have you previously set the time via NTP and set a time zone?

DmitryAVR, 2022-07-27:
For example. Set the timer as in the screenshot. Charging on pause. Time zone: Etc/Universal
[image]

Next, I set the timer to start 3 hours earlier and the charging was activated.
[image]

closed 2022-09-27 by DmitryAVR