OpenEVSE/openevse_esp32_firmware #393: Captive portal broken in V4.1.3

glynhudson, 2022-07-21:
The android captive portal doesn't seem to work in V4.1.3, the interface starts to load the captive portal closes. Connecting directly to the unit using a laptop that does not support captive portal allows the interface to load fine. See video:

https://user-images.githubusercontent.com/758844/180216068-ec78ab3c-7644-4b07-b38d-fb2d47369049.mp4

This is a significant issue since most users will use a mobile device and captive portal to setup the WiFi connection. I've marked V4.1.3 as a pre-release.

jeremypoulter, 2022-07-23:
@glynhudson Can you check the device logs? Is it rebooting?

jeremypoulter, 2022-07-24:
@glynhudson Can you confirm which device you where using? I have found an issue with ESP32 Gateway I am looking in to, but a normal WiFi only board looks to work fine:

https://photos.app.goo.gl/3hBJq53YCGmk3rxs5

glynhudson, 2022-07-25:
I was using the normal OpenEVSE WiFi V1 board

glynhudson, 2022-07-25:
The modules is not rebooting, here's the debug output while I was trying to load a captive portal:

[image]

The captive portal is strange, after lots of attempts I finally got it to load, but they it wouldn't load the standalone interface, see video:

 https://drive.google.com/file/d/1634k6GlCrrtwr8nLgN7JE0XugOEIsEmn/view?usp=sharing

glynhudson, 2022-07-25:
Update: I've tested using a couple of different Android phones and the captive portal works fine :+1: 

There must be some caching issue on my phone, since I had just previously connected to older versions of the WiFi firmware. 

Mmm even after clearing the Android Chrome cache on my phone, I'm still having issues getting the portal to stay open. However, since I can't recreate the issues on my colleagues phones the issue must be with my phone.

closed 2022-07-25 by glynhudson