OpenEVSE/openevse_esp32_firmware #399: Does not report Wh to MQTT

andriuskr, 2022-07-30:
While charging, Wh topic is not updated to MQTT. It is alway the same value of 191895
. I assum it should represent charging power.
[image]
[image]

jeremypoulter, 2022-07-30:
This is the same as the `total_energy` topic, it is only updated at the end of the charging session.

andriuskr, 2022-07-30:
Leaving no way to get current power usage ? :(

jeremypoulter, 2022-07-30:
The session energy is available as `session_energy`, there is no instantaneous power, but both the current and the voltage are available for a simplistic power calculation

KipK, 2022-07-30:
One could add a real time total counter, the total can't be used directly for counting consumtion prices. I had to make my own externally.

Having a real time counter has sense for external tools.

jeremypoulter, 2022-08-01:
Related to #224

closed 2023-03-30 by glynhudson