OpenEVSE/openevse_esp32_firmware #400: web UI localisation

KipK, 2022-07-30:
I have started add localisation the web UI ( english / french), using i18next-ko npm.

The WIP branches are here:

https://github.com/KipK/ESP32_WiFi_V4.x/tree/localisation
https://github.com/KipK/openevse_wifi_gui/tree/i18next-ko

If you have any suggestions while I'm on it.
Works well for now, but a pain to complete :)

KipK, 2022-07-31:
It's mostly ready. As it's based on Current Shaper branch, I will pull request this after the shaper ,merge.

jeremypoulter, 2022-08-01:
Great work, this is something that has been missing for a while now, hopefully we get some more contributions for other languages now too.

Knockout.js is a means to an end, and a vast implement over what we had before ;-). Ultimately we need to rebuild the UI using more modern tools, probably based on Material UI.

KipK, 2022-08-01:
I'd vote for a Svelte.js or Preact js fw with any light CSS fw like Purecss or Skeleton.
Material is a bit heavy but I think we can limitate that with tree-shaking.

Also first, UX should be revisited a bit, actual thing looks more like a patchwork of addition :)

jeremypoulter, 2022-08-01:
Indeed the current UI is a patchwork of stuff that has evolved over the lifetime of the project. I have created #401 to track the requirements/ideas for the new UI.

jeremypoulter, 2022-09-24:
Fixed via #422

closed 2022-09-24 by jeremypoulter