OpenEVSE/openevse_esp32_firmware #404: Error from emoncms

cholawo, 2022-08-02:
I'm running the latest development release. The "Energy Monitoring" area in OpenEVSE says it's not connected.   I have an error coming back from emoncms when data is sent to it.   Here is what I see in the OpenEVSE serial console debug:

`{"amp":0,"voltage":244.9400024,"pilot":6,"wh":2017162,"session_energy":0.258333333,"total_energy":2017.162,"temp":300,"temp_max":300,"temp1":false,"temp2":300,"temp3":false,"temp4":495,"state":254,"flags":1280,"vehicle":1,"colour":6,"manual_override":0,"override":"{\"state\":\"null\"}","freeram":163232,"divertmode":1,"srssi":-71}
{"emoncms_connected":0,"emoncms_message":"Input in not a valid JSON object"}`

I would expect the error is caused by this: `"override":"{\"state\":\"null\"}"`

jeremypoulter, 2022-08-02:
Technically this is an EmonCMS bug... It is valid JSON, just not a number. Need to update to only send numbers to EmonCMS as it doesn't know what to do with strings.

cholawo, 2022-08-04:
Yes you're right, I see this in the error log of emoncms:

```
2022-08-04 15:15:33.678|ERROR|input_controller.php|{"success": false, "message": "Input in not a valid JSON object"} for User: 1
2022-08-04 15:15:33.816|ERROR|emoncms_mqtt.php|ErrorException: A non-numeric value encountered in /var/www/emoncms/Modules/process/process_processlist.php:829
Stack trace:
#0 /var/www/emoncms/Modules/process/process_processlist.php(829): exceptions_error_handler(2, 'A non-numeric v...', '/var/www/emoncm...', 829, Array)
```
So are you saying this won't be fixed until someone updates ESP32_WiFi_V4.x to stop attempting to send numbers to EmonCMS?

jeremypoulter, 2022-08-24:
Fixed by #416

closed 2022-08-24 by jeremypoulter