OpenEVSE/openevse_esp32_firmware #407: Olimex ESP32-PoE-ISO Ethernet failing to start up

ianrrees, 2022-08-03:
Hello,

      I'm trying to set up a wired OpenEVSE, using an Olimex ESP32-PoE-ISO dev board that I had on hand - it looks functionally identical to the ESP32-GATEWAY used in the ethernet documentation here.  The latest ESP32 WiFi firmware here seems to work fine (just testing this on the bench, not hooked up to OpenEVSE hardware yet) but regardless of whether the ethernet is connected, the WiFi AP starts up.  No link/act lights on either the ethernet switch or device end.

On the serial console, I see "E (2014) lan87xx: lan87xx_init(499): wrong chip ID" - which looks suspiciously similar to [this issue](https://github.com/espressif/arduino-esp32/issues/6142).

PlatformIO is new to me, and grepping around, I don't see any reference to where the lan87xx driver might come from.  Just hoping that someone could help point me in the right direction, thanks!

ianrrees, 2022-08-03:
Minor update: I stuck a 1s delay before `ETH.begin();` and that doesn't change the symptoms at all, so perhaps it's more complicated.  Will aim to dig in further after work today.

ianrrees, 2022-08-05:
Got that issue sorted - I hadn't noticed these two boards use a different GPIO pin to switch the power to the ethernet PHY.  Will issue a PR once I'm a little further along.

closed 2022-08-05 by ianrrees

jeremypoulter, 2022-08-05:
Looking forward to seeing it.

ianrrees, 2022-08-07:
Here's my work-in-progress, it has some hacky changes and certainly won't be PRed like this: https://github.com/ianrrees/ESP32_WiFi_V4.x/tree/esp32-poe-iso

vmuukka, 2022-08-17:
Add 2 sec delay in the start of program. That will fix the ethernet connection.

Informaticore, 2023-06-21:
@ianrrees do you have any news? I am also running the ESP32-PoE-ISO Board but I am receiving the same issue. 
The Gateway from Olimex does work! 

How is the PHY_PWR used here? This is default enabled so you should not really need to touch it, right? But both boards using the same LAN7810A Chip but for the ISO Board I am receiving the "Chip ID wrong" error.

Any idea whats the issue here?

ianrrees, 2023-06-21:
Hi @Informaticore , the issue was fixed with a change to the GPIO configuration for the ESP32-PoE-ISO, and this item was closed because the change were pulled in to the main branch.  Sorry, but it doesn't look like the CI is set up to release firmware images for that board - I think that there is still some weirdness with ethernet-connected OpenEVSE so perhaps that is why?

The relevant changes were in the files `src/net_manager.cpp` and `platformio.ini` - you can see them in [the branch referenced above](https://github.com/ianrrees/ESP32_WiFi_V4.x/commit/54ffeac018c05a8e77c6dbfe7d70c009b935600d).  They add support for building firmware for ESP32-PoE-ISO using `pio run -e openevse_esp32-poe-iso` (perhaps with additional arguments per your situation).  If you encounter problems building firmware, please have a look through the documentation and if that doesn't work, create a new issue so we can try to help.

Informaticore, 2023-06-21:
@ianrrees hey no no I am actually having trouble with a different firmware using Toit but the issue is the exact same and I was wondering if can fiy it using your changes there.

ianrrees, 2023-06-21:
Not sure how I can add more to the previous response - you'll need to look at the (small) changes and decide if they apply to your situation.