OpenEVSE/openevse_esp32_firmware #408: Feature Request: Start/End Charge timestamp

firstof9, 2022-08-05:
I would love to see a timestamp when the current charge session has started then an end timestamp made available via websocket/mqtt UTC time.

`start_charge: <timestamp>` 
`end_charge: <timestamp>`

When charge starts `end_charge` could be `null`.

jeremypoulter, 2022-08-06:
That information is available in the logs in a relatively machine friendly manor, but I assume you mean more that it should just be as a part of the status? 

Also what would you class as the start/end of the charge? You may think that is obvious but don't forget some modes, like PV divert, can start/stop the charging between the vehicle being plugged in and being fully charged. The session energy meter counts the energy between the car being plugged in and disconnected so maybe the time stamps should align to that? But equally if you have the car set to off peek charge via the timer I would imagine it could take multiple days for cars with large batteries to charge. Would you count each one as a separate session?

firstof9, 2022-08-06:
> Also what would you class as the start/end of the charge?

Yes at the session start/end.

> just be as a part of the status?

Yes, digestible via the websocket would be great.

> Would you count each one as a separate session?

Just the session start/end that's already used would be fine.

closed 2026-09-21 by github-actions[bot]