OpenEVSE/openevse_esp32_firmware #409: GUI should notify when firmware reboots

ianrrees, 2022-08-07:
While hacking on ESP32 firmware today, I noticed in the serial console that the watchdog was firing and resetting the ESP32, but there was no notification on the web GUI.  I understand that this is not a normal use-case, but it would be good if there were at least a notification in the GUI when something like this happens.

Maybe something that can be done as part of #401 ?

glynhudson, 2022-09-13:
I'm not sure this is needed, best to fix the issue to ensure the controller does not reboot. I've not been able to recreate.

closed 2022-09-13 by glynhudson

ianrrees, 2022-09-13:
Hi @glynhudson - I agree it's not strictly necessary, but actually you can't ensure that the controller does not reboot.  For instance, the power supply could be failing; so the question is how to respond when that reboot happens.  I think it's fair to say that the current implementation implicitly hides these fault conditions, and in my opinion that's not great.

I've found that it's quite handy in troubleshooting and developing to have some user-observable notification when a system fails, and if it's done right that functionality won't get in the way.  Just as a thought: if the status JSON had an "uptime" then the browser code could quite easily make an appropriate notification.