OpenEVSE/openevse_esp32_firmware #414: bug/crash with scheduler api

KipK, 2022-08-10:
I'm in holiday I can't track it but seems I saw a bug if only one schedule is set ( for example a "disabled" schedule with id: 1 , without an "enabled" one ) 

Can't see if it's a crash or bug from here, but then common status data from create_rapi_json are not published on MQTT, I think it's the same with http,  event_send is not called. Either it crashes before or something else.

jeremypoulter, 2022-08-10:
I did have an issue with something similar, thought I fixed it, will have a look.

KipK, 2022-08-31:
I see some events.getnext() in scheduler.cpp without checking if there's such next event first, could this be the culprit ?

KipK, 2022-10-10:
While working on the Timers part of the new GUI, I get this issue as the UX I'm doing is creating each timer separated.

I confirm the OpenEvse-wifi crash if there's only one timer. 
Module crash in the wollowing 30 secs.

It doesn't happend if there's more, whatever the enable/disable parity is respected or not.

jeremypoulter, 2022-11-12:
Sorry, I have not had a chance to look at it yet

KipK, 2023-01-28:
Just a hint as I'm looking at it, it's not the fact that we have only one timer, it seems to need booth an active and disabled timer. I've tried with 2 enabled or disabled timers, and it also crash the firmware.
But I still don't know where it segfault, I need a stacktrace

PR #549 (2023-02-22): Fix for infinate loop in getNextEvent
getNextEvent could get stuck in an infinate loop if searching for a type that is not in the list. This can happen when there is only enteries ofa single type, eg disabled and you search for active, as the LCD module does causing a watchdog timeout.

Fixes #414
commit 169c7893 (2023-02-22): Fix for infinate loop in getNextEvent

getNextEvent could get stuck in an infinate loop if searching for a type that is not in the list. This can happen when there is only enteries ofa single type, eg disabled and you search for active, as the LCD module does causing a watchdog timeout.

Fixes #414
closed 2023-02-22 by glynhudson