OpenEVSE/openevse_esp32_firmware #424: MQTT client does not subscribe to all topics - mqtt_loop() missing?

acobo, 2022-09-01:
Hi,
 I haven't been able to access the MQTT API using the <base_topic>/claim/set  topic.  I have found in my mqtt server (mosquitto) logs that the client does not subscribe to all topics, only the first four listed in the source code:

` 
 // MQTT Topic to subscribe to receive RAPI commands via MQTT
    String mqtt_sub_topic = mqtt_topic + "/rapi/in/#";

    // e.g to set current to 13A: <base-topic>/rapi/in/$SC 13
    mqttclient.subscribe(mqtt_sub_topic);

    // subscribe to solar PV / grid_ie MQTT feeds
    if(config_divert_enabled())
    {
      if (mqtt_solar != "") {
        mqttclient.subscribe(mqtt_solar);
      }
      if (mqtt_grid_ie != "") {
        mqttclient.subscribe(mqtt_grid_ie);
      }
    }
    // subscribe to current shaper MQTT feeds
    if(config_current_shaper_enabled())
    {
      if (mqtt_live_pwr != "") {
        mqttclient.subscribe(mqtt_live_pwr);
      }
    }
    // subscribe to vehicle information from MQTT if we are configured for it
    if (mqtt_vehicle_soc != "") {
        mqttclient.subscribe(mqtt_vehicle_soc);
    }
    if (mqtt_vehicle_range != "") {
        mqttclient.subscribe(mqtt_vehicle_range);
    }
    if (mqtt_vehicle_eta != "") {
        mqttclient.subscribe(mqtt_vehicle_eta);
    }

    if (mqtt_vrms!="") {
      mqttclient.subscribe(mqtt_vrms);
    }
    // settable mqtt topics
    mqtt_sub_topic = mqtt_topic + "/divertmode/set";
    mqttclient.subscribe(mqtt_sub_topic);

    mqtt_sub_topic = mqtt_topic + "/shaper/set";
    mqttclient.subscribe(mqtt_sub_topic);

    mqtt_sub_topic = mqtt_topic + "/override/set";        
    mqttclient.subscribe(mqtt_sub_topic);

    mqtt_sub_topic = mqtt_topic + "/claim/set";        
    mqttclient.subscribe(mqtt_sub_topic);
`

mqtt_vrms is the last topic subscribed, and I cannot see the reason why the next four subscribe() are not reached. 

Incidentally, I found the very same problem with an Arduino project several years ago, using the PubSubClient library. The reason then was that the background process of the mqtt client library was not able to attend more than 4 or 5 subscribe() calls in a row. A call to the loop() function of the mqtt library after each subscribe() call solved this issue.

I have tried to add a call to mqtt_loop()  after each subscribe() in mqtt.cpp, and it solves the problem.

I don't know if this a specific issue with my wifi setup, but maybe it won't hurt to add those mqtt_loop() calls just in case.

Thankyou!
 Adolfo.

KipK, 2022-09-06:
Hi

I haven't got this problem here, but if it solves yours I'll add a temp on the subscribing

KipK, 2022-09-06:
Can you try just adding a yield() instead of mqtt_loop() ?

acobo, 2022-09-06:
Hi,
 I have tried with yield() and it works for me also, thankyou!

KipK, 2022-09-06:
Thanks for testing. :) I'm gonna submit a PR

PR #425 (2022-09-07): fix #424
add yield() after mqtt subscribe
this fix #424
commit 6b8319bb (2022-09-07): Merge pull request #425 from KipK/mqtt

fix #424