OpenEVSE/openevse_esp32_firmware #428: V4.1.4 does not work with ESP gateway

glynhudson, 2022-09-10:
ESP gateway Ethernet Module seems to boot up, then crash with error `analogRead(): Pin 1 is not ADC pin!`

V4.1.3 works fine 

Testing on rev G hardware

```
OpenEVSE WiFI 7294
Firmware: 4.1.4
Build date: Jul 21 2022 12:11:28
IDF version: v4.4.1-1-gb8050b365e
Free: 270496
[   225][E][vfs_api.cpp:104] open(): /littlefs/schedule.json does not exist, no permits for creation
[   226][E][esp32-hal-adc.c:180] __analogRead(): Pin 1 is not ADC pin!
AP IP Address: 192.168.4.1
E (10412) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10412) task_wdt:  - loopTask (CPU 1)
E (10412) task_wdt: Tasks currently running:
E (10412) task_wdt: CPU 0: IDLE
E (10412) task_wdt: CPU 1: IDLE
E (10412) task_wdt: Aborting.

abort() was called at PC 0x4012d7f4 on core 0

Backtrace:0x400836e9:0x3ffbed8c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:792
load:0x40078000,len:8160
ho 0 tail 12 room 4
load:0x40080400,len:5732
entry 0x4008069c
```

jeremypoulter, 2022-10-05:
@glynhudson I have not been able to reproduce this, can you try removing the `RESET_ETH_PHY_ON_BOOT=1` this may break the rev G+ boards, I only have an E to test.

glynhudson, 2022-10-13:
Great, removing `RESET_ETH_PHY_ON_BOOT=1` has worked on rev G. Shall I open a PR to remove this, or do we need to create a new separate pio environment for rev F/G? Are there any detrimental effects of removing this?

jeremypoulter, 2022-10-13:
Think it is needed for the E board that I have so many need a new env for F and G. Can you also test how stable the ethernet is after reboot of the device (not a power cycle)?

glynhudson, 2022-11-03:
Fixed https://github.com/OpenEVSE/ESP32_WiFi_V4.x/commit/b82188a2fa04bbdf14155433d8b9945359311f0d

closed 2022-11-03 by glynhudson