OpenEVSE/openevse_esp32_firmware #429: Delay timer & manual override don't work on latest master branch, unable to start charge

glynhudson, 2022-09-13:
I've just been testing the latest master branch, and I've noticed the delay timer and manual override don't work. It's impossible to start a charge. The only way to get a charge going is by restarting the controller:

[image]

[OpenEVSE.webm](https://user-images.githubusercontent.com/758844/189783872-1c6ee6b2-64b5-4e0b-9855-864eb5605fac.webm)

I've looked through the recent commits to see what could have caused this, the only changes to the scheduler I can see were made by @KipK

KipK, 2022-09-14:
Hi @glynhudson 
Unfortunately I can't reproduce here. It works as expected.
Are you sure you have updated the gui ( it probably needs a clean from the gui/dist directory before npm run build

Here is my build for openevse_wifi_v1 from current Master, can you try it ? 

https://drive.google.com/file/d/1AR_a9inA52eZOQv8_6SUzpNSiqfh6mBv/view?usp=sharing

glynhudson, 2022-09-19:
Mm I seem to have the same issue with your build and V4.1.4. However, V4.1.3 works fine. I'm sure V4.1.4 worked well previously therefore there's a good chance this issue is with my setup. I'll keep doing some testing..

closed 2022-09-19 by glynhudson

KipK, 2022-09-20:
I'd like to see some debug logs from your side to give a look if you can post some.

glynhudson, 2022-10-12:
Finally, I've managed to resolve this issue with the help of @jeremypoulter. 

It turns out that RFID was enabled which was blocking the change. My system doesn't have any RFID hardware, I must have enabled it while I was testing and forgot to disable. Totally my fault, but this has highlighted the importance of displaying a status message to the user to notify what is causing a charge to be blocked. I've created an issue: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/442

Also, if no RFID hardware is present it shouldn't be possible to enable RFID: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/427