OpenEVSE/openevse_esp32_firmware #432: The delay timer does not work correctly when you set the time via a browser or manually

DmitryAVR, 2022-09-27:
When you reboot EVSE, EVSE enters the sleeping state. But the current time is included in the timer interval

jeremypoulter, 2022-09-27:
Is this not the issue raised in #386?

DmitryAVR, 2022-09-27:
No. The main thing there is the wrong time, which is set via the browser. I closed that question. Here, as I understand it, the timer does not work correctly when restarting, because there is no event "Scheduler woke: WakeReason_Event". 

[image]

But this event is in NTP. How can I enable this event when setting the time via the browser?

glynhudson, 2023-03-31:
This should be fixed in the latest version

closed 2023-03-31 by glynhudson