OpenEVSE/openevse_esp32_firmware #433: Feature Request: Minimum Idle Time

pdhoogh, 2022-09-27:
Concerning OpenEVSE Eco mode and Current Shaper.

For the purpose of reducing wear on the contactor and in light of legislation trying to avoid peak grid load, the proposal is to have a _Minimum Idle Time_ that could be used instead of the _Minimum Charge Time_.

With the current available functionality using _Minimum Charge Time_ , when while charging the EV another big power load kicks in and the minimum charge time has not yet run out, the OpenEVSE keeps charging and a peak load of over 5 kW may occur. If that coincides with the sun being obscured by a cloud, the peak power can go even higher than 8kW.

With the new proposed feature _Minimum Idle Time_ used instead, the charging of the EV would stop immediately after a grid power import surge develops, helping to reduce the power peak. Once charging was interrupted, the EV charge would not resume before the _Minimum Idle Time_ timer has expired, helping to reduce wear on the contactor.

The UK is putting in legislation for EVSE behavior to counter grid power import peaks. In Belgium the electricity distribution tariff will, as of January 1st 2023,  add a load peak component. The smart meter registers the load in 15 minute chunks. If one 15 minute chunk shows 8kW continuous, that one 8kW peak is used to make you pay for a 8kW consumption during the whole month. 

This proposal is for trying to help reduce grid import peaks while still reducing wear on the contactor.

soloam, 2022-11-05:
I was going to open this feature request! On my case the shaper is constantly turning on and off the relay, when the current goes below 6. I ended up doing a automation on Home Assistant to avoid this, but it's not as effective as if it would be in the actual openevse.

As requested, if charger turns off, it should remain idle until the current is again over 6 for a minimum amount of time. For example :

Let's assume a idle time of 10 minutes. 

10:00 - Goes bellow 6 - System goes idle
10:30 - Remains bellow 6 - Remains Idle
10:35 - Goes to 10 - Remains idle for 10 min
10:40 - Goes bellow 6 - Remains idle and reset timer
10:50 - Goes to 10 - Remains idle for 10 min
11:00 - Remains over 6 - Resumes Charging

Thank you

pdhoogh, 2022-11-05:
Your proposal is slightly different from what my intention was, but your proposal has merit.

The original thought was to save the relay and avoid import power peaks. So as soon as the relay is OFF, a e.g. 10 minute timer would start and only if it runs out can OpenEVSE close the relay again. As soon as the available power would drop below 6 Amps, the relay opens and the 10 minute timer starts.

With cyclic loads like a dryer, that may lead to OpenEVSE closing the relay every 10 minutes for only 30 seconds because the dryer causes the available current to drop below 6 Amps every time it turns on the heating element. With your proposal this would not happen. OpenEVSE would wait for 10 minutes after the dryer is done to engage the EV charge again and would keep the relay off all the time that the dryer is cycling its heating element. So your proposal is even better for reducing wear on the relay.

Of course one can buy a heat pumped dryer to avoid this specific scenario... it would be interesting to have a label on OpenEVSE "Only works with heat pump dryers" :-)... the good thing is that your proposal also works better with slightly clouded skies where the PVs are cycling production all the time. Good thinking!

soloam, 2022-11-05:
I ended up doing something similar on a Home Assistant automation! To try to spare a relay! But this would be a lot better if done natively on openev. Hope that this gets some attention.

pdhoogh, 2022-11-05:
I am confident it will get done.

Another way of doing this may be using a Python program to disable OpenEVSE through MQTT for the required time and enabling it again after that time. I've done this, it works too.

I agree doing it "native" would be better as one never knows how doing those disable / enable cycles messes up the OpenEVSE operation even further.

jeremypoulter, 2022-11-05:
The ideal time was specifically for the divert, the current shaper is much more of a safety feature so won't have the ideal timer,  @soloam I think it would be best to raise a new ticket describing what your use case/goal is.