OpenEVSE/openevse_esp32_firmware #435: Unstable env:openevse_nodemcu-32s

DrFrankReade, 2022-09-28:
Getting instability, I suspect that it's the non-definition of defaults for buttons and improper or non-use of pullup/down on the many flavors of nodemcu-32s boards out there. 

Suggest defining the defaults in platformio.ini  so that **dummies like me** know to at least change them. 

```
[env:openevse_nodemcu-32s]
board = nodemcu-32s
build_flags =
  ${common.build_flags}
  ${common.version}.dev
  ${common.src_build_flags}
  ${common.debug_flags}
  -D WIFI_LED=2
  -D WIFI_LED_ON_STATE=LOW
  -D RAPI_PORT=Serial2
  -D DEBUG_PORT=Serial
  -D WIFI_BUTTON=0
  -D WIFI_BUTTON_PRESSED_STATE=LOW
```

jeremypoulter, 2022-09-28:
Thanks for the input, if you have working settings for this board then a PR would be great, however there may be different versions of the board so can you clarify the source/revision of your board?

DrFrankReade, 2022-09-29:
Unfortunately, the board is so desperately generic that it goes by no other name, as far as I can tell, than the NODEMCU-32s

The best way I can differentiate it from others is by the position of the yellow capacitor and the voltage regulator. 

[image]
```
[env:openevse_nodemcu-32s]
board = nodemcu-32s
build_flags =
  ${common.build_flags}
  ${common.version}.dev
  ${common.src_build_flags}
  ${common.debug_flags}
  -D WIFI_LED=2
  -D WIFI_LED_ON_STATE=LOW
  -D RAPI_PORT=Serial2
  -D WIFI_BUTTON=0
  -D WIFI_BUTTON_PRESSED_STATE=LOW
```
The above settings have been fairly stable over the last 12 hours, which is better, but hardly saying much.

DrFrankReade, 2022-11-02:
Update: Experiencing the same on a brand new "official" OpenEVSEWifi board (came with 4.1.2) and broke with 4.1.5. Suspect it's not hardware.

closed 2022-11-03 by glynhudson