OpenEVSE/openevse_esp32_firmware #436: Enhancement Request: Remote Restart Functions

davethetallguy, 2022-09-28:
Enable the "Restart OpenEVSE WiFi" and "Restart Open EVSE" functions, currently implemented as buttons on the Advanced OpenEVSE page, via remote (http) access.

Use case: one of the sub-functions (e.g. MQTT, EmonCMS, <future function>) is misbehaving, but the EVSE is still accepting input. This would permit remote restart of the device via Home Automation (e.g. [Home Assistant](https://github.com/firstof9/openevse)) or similar to reset the sub-function completely.

jeremypoulter, 2022-09-28:
All actions in the UI are accessible via the HTTP API, in this case `/restart` for the WiFi and by sending the restart RAPI command for the EVSE module (not sure what that is off the top of my head).

As these are not documented, the easiest way to see what is needed is to use your browser's Dev Tools and look at what is sent when you click the buttons.

That being said sending RAPI commands via HTTP (or MQTT) is being removed so we will implement an equivalent 'REST' API for restarting the EVSE and when we do this the `/reset` endpoint may change to avoid confusion, but that is TBD, and I guess we can use this ticket for those changes...

davethetallguy, 2022-09-28:
Apologies... I actually meant to request a REST implementation, although I didn't know the http command is available currently. Maybe I can rig up a workaround using HTTP until the REST command is implemented, and WiFi is generally what I need to restart.  Thanks.

firstof9, 2022-09-30:
I'll work on a `restart` button for Home Assistant :+1:

closed 2022-10-12 by glynhudson