OpenEVSE/openevse_esp32_firmware #439: API feature req: update external value from http API

KipK, 2022-10-05:
I've received this request from another forum. 

We should be able to update external values like Voltage, Vehicule SOC, +I/-E and co from HTTP POST requests and not only MQTT.

Sattaz, 2022-10-05:
I'm the guy from that other forum and I developped a plugin for the Jeedom home automation system using the EVSE RAPI.
I was adapting my plugin to use the new 'WIFI' API but could not found any http/post method to update such settings as the Voltage, etc ... as referred by KipK.
This is essential to get the EVSE module precisely doing internal calculation like the charge session.
Could you implement this? Thx a lot!

jeremypoulter, 2022-10-05:
I guess we could do this, similar to the config API I could allow a `POST` on the `/status` endpoint to update those values that can be updated by MQTT.

jeremypoulter, 2022-10-05:
Also if Web Sockets are an option I am also planning on adding support for reading from EmonESP #304.

fhteagle, 2022-10-15:
+1 for this, especially being able to control the max amps value via HTTP post or get.

Any security considerations with enabling this though?

KipK, 2022-10-15:
> 

You already can set max_current using claims or override from the API

commit 39a10b9d (2022-10-24): Merge pull request #453 from KipK/poststatus

fix #439  added POST /states
closed 2022-10-24 by jeremypoulter