OpenEVSE/openevse_esp32_firmware #440: Attempting to add bigger LCD

prathamesh-gharat, 2022-10-08:
I am trying to add a bigger LCD but connect directly to the ESP32 instead. Has any such attempt been made in the past? Are there any possible issues that someone can foresee?

jeremypoulter, 2022-10-08:
This is not something we have done yet, there are a lot of assumptions that it is a 16x2 display currently, but most of the display logic is abstracted to the LCD class/file

prathamesh-gharat, 2022-10-08:
I am trying to understand how it works. I am a bit new to C++ (the last I did anything related to C++ was 5+ years ago; been on the web side of things so far; JS+PHP). So I am just trying to pick up and follow the existing patterns. I have tried the following things so far.

1. Added u8g2 to dependencies in platform.io `lib_deps = olikraus/U8g2@^2.34.3`
2. I created a new header and cpp file named `big_lcd.h` and `big_lcd.cpp` with just a `begin` and `lcd_loop` method.
3. Took a working (i.e. tested & found working individually on the ESP32 and the LCD) example code from Arduino and tried to get it running from within the OpenEvse WIFI gateway code, it compiles and gets successfully uploaded to the ESP32, but it does not produce any output on the LCD.

Code just for reference.

big_lcd.cpp
```cpp
#include "emonesp.h"
#include "big_lcd.h"
#include "RapiSender.h"
#include "openevse.h"
#include "input.h"
#include "app_config.h"
#include <sys/time.h>

#include <Arduino.h>
#include <U8g2lib.h>

#include "evse_man.h"
#include "scheduler.h"
#include "manual.h"

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define LCD_RESET 4   // RST on LCD
#define LCD_CS 5      // RS on LCD
#define LCD_CLOCK 18  // E on LCD
#define LCD_DATA 23   // R/W on LCD

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, /* clock=*/ LCD_CLOCK, /* data=*/ LCD_DATA, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);

static void IGNORE(int ret) {
}

BigLcdTask::BigLcdTask() :
  MicroTasks::Task()
{
}

void BigLcdTask::begin(/*EvseManager &evse, Scheduler &scheduler, ManualOverride &manual*/)
{
  MicroTask.startTask(this);

  u8g2.setBusClock(950000);
  u8g2.begin();
}

void BigLcdTask::setup()
{

}

unsigned long BigLcdTask::loop(MicroTasks::WakeReason reason)
{

}

void BigLcdTask::lcd_loop() {
  u8g2.clearBuffer(); // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer(); // transfer internal memory to the display
  delay(1000);
}

BigLcdTask big_lcd;
```

Changes to main.cpp (diff)
```diff
void setup()
{
...
  lcd.begin(evse, scheduler, manual);
++  big_lcd.begin();
#if defined(ENABLE_PN532)
...
void
loop() {
...
  divert_current_loop();

++  big_lcd.lcd_loop();

  Profile_Start(MicroTask);
```

The code that works when compiled and uploaded through the Arduino IDE to the ESP32.
```cpp
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define LCD_RESET 4   // RST on LCD
#define LCD_CS 5      // RS on LCD
#define LCD_CLOCK 18  // E on LCD
#define LCD_DATA 23   // R/W on LCD

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R2, /* clock=*/ LCD_CLOCK, /* data=*/ LCD_DATA, /* CS=*/ LCD_CS, /* reset=*/ LCD_RESET);

void setup(void) {
  u8g2.setBusClock(950000);
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}
```

Here are a few of the possible issues I can think of.

1. My lack of knowledge of C++ or this project.
2. The pins specified for the LCD might be used by something I am not aware of.

prathamesh-gharat, 2022-10-09:
I added a few `DEBUG.println` to make sure the `begin` and `lcd_loop` functions are being called. Used the serial monitor to confirm those messages, and they do appear.

jeremypoulter, 2022-10-09:
Your assumption is correct the LCD driver is on the EVSE module and the WiFi module sends the commands to display the text to it.

You shouldn't need any additional loop, you should add your loop code to `BigLcdTask::loop` and instead of calling `delay` you should return the required delay from `BigLcdTask::loop` 

Which board are you using?

prathamesh-gharat, 2022-10-09:
I switched one ping from GPIO 4 to 22 and it started working.

> You shouldn't need any additional loop, you should add your loop code to BigLcdTask::loop, and instead of calling delay, you should return the required delay from BigLcdTask::loop

Thanks! I will look a bit more into `MicroTasks`.

> Which board are you using?

I am using an ESP32-WROOM32 dev board for testing.
I made the following change earlier while trying out things. 
```diff
--board = esp32-gateway
++board = esp32dev
```

I'll try to rewrite parts of the LCD code for the BIG_LCD.

closed 2022-10-09 by prathamesh-gharat

jeremypoulter, 2022-10-09:
There is a env specifically for the ESP32-WROVER32 dev board, `env:openevse_esp-wrover-kit `

1madison24, 2023-12-13:
I am currently working within this openevse_esp32_firmware espressif_esp_wrover-kit_latest environment and the lcd code seems to not work on my board. I have tried two different wrover kits and neither of them appear to have anything on display. Is there anything I must add to the code to get the lcd on the wrover kit to work and appear like it does to simulate the openevse hardware?