OpenEVSE/openevse_esp32_firmware #442: RFID authentication user notification

glynhudson, 2022-10-12:
If RFID is enabled causing charging session to be blocked awaiting authentication a status message should be displayed in the UI

KipK, 2022-11-18:
Good thing is it's naturally handled on the new UI. 

[image]

I've added an exception for RFID to only display one Button ( Start Charge, Stop Charge ) and No Auto button, as RFID has higher priority than manual override, there's no way to force manual mode or set an auto mode.

So behavior for RFID is, if no badge, button wait, uncliquable, , if authenticated, then Display button start charge, if charging, display button stop charge. 

I

jeremypoulter, 2022-11-18:
Once authenticated the charger should operate as normal I believe, the RFID claim should be removed letting any other mode do its thing... Or at least that was the intent, fairly new feature so may have bugs

KipK, 2022-11-18:
you're right, I've jsut checked, and RFID release it's Disabled claim at authentification.
So other services grabs the priority.
Things brings us an interesting question on how do we want the UI to behaves for RFID.

Either classical like, like the other services, so we display the On/Auto/off buttons ( disabled first when not authenticated yet ) 

Or we choose to remove Auto to just propose a one button Start / Pause charge. 
Why such a difference ? Because while using RFID, the common use case I think is to authenticate for one charge session. 
Then there's no reason to display the Auto Release switch, nor Shaper / Divert / Schedule mode there, or won't it deserve the fact we need authentification there ? 

That makes me think we should also move the Scheduler part in the Configuration section too. There's no need to have it configurable in Main page.

jeremypoulter, 2022-11-18:
For now I think the RFID should just be used as an unlock, IE you plug the car in then use the RFID to unlock the EVSE. After that it should behave as it would with no RFID.

Even in the UI I wouldn't change anything, just the note to say disabled waiting on RFID, all other operations should be available irrespective of the RFID state. I think this is the most appropriate for home use. Any commercial use can be dealt with when we have more specific requirements, but it will be a custom build in all likelihood anyway.

KipK, 2022-11-18:
ok, so we're going to something like this: 

[image]

then once connected back to normal:

[image]

Perhaps I should add a tag to display that rfid is authenticated ( displaying badge ID ? )

closed 2023-03-30 by glynhudson