OpenEVSE/openevse_esp32_firmware #450: 4.1.5 seems to have issues in eco mode

pdhoogh, 2022-10-19:
4.1.4 had issues that were documented and Jeremy Poulter had some fixes in patch releases for 4.1.4. I installed and tested those. They seemed to work OK with PV Power Ratio = 1. No complaints. 

Then 4.1.5 was built as the latest release and I installed it. Soon after the OpenEVSE started misbehaving.

The day before yesterday, at night, I plugged in the EV. OpenEVSE was in eco mode. There was no solar. The patched 4.1.4 worked correctly: it would wait for sufficient solar next morning and start charging the EV with throttled current. Not 4.1.5 so it seems. It started charging the EV and let current go all the way to max (set 15A) without solar. I tried a number of different things like resetting the WiFi, resetting OpenEVSE, power cycle, turning override on/off several times... nothing worked, OpenEVSE kept charging. WiFi was working and Power1 was updated correctly, but I noticed that the double _Smoothed Current_ indicators on the GUI were not updating.

I waited till today to try again during a sunny day. Behavior was erratic. I was not able to find a systematic (mis-)behavior. as the plots below show, sometimes it would throttle correctly, sometimes it gives up and moves the charge current to max. Look at Power1 as the best indicator for the OpenEVSE intermittently not throttling:
[image]

I will move the WiFi version back to Jeremy's patched 4.1.4 and see what happens.

pbix, 2022-10-23:
After upgrading to 4.1.5 I was unable to save settings.  There was no error message, my changes just disappeared at the next screen refresh.

I also did a factory reset.  So far my changes are not being saved.

jeremypoulter, 2022-10-24:
When it is going wrong can you post the values of the following endpoints:
- `/claims`
- `/claims/target`
- `/status`
- `/r/$GC`

pdhoogh, 2022-10-26:
Hi Jeremy,

Away from home for a while. Will take a few weeks before I can get to this.

———————————
Philippe D’Hooghe
Meidoornlaan 10
9820 Merelbeke
+32 485 655 669

closed 2023-04-03 by glynhudson