OpenEVSE/openevse_esp32_firmware #451: Feature Request / Planning : Bidirectional AC charging via J1772

fhteagle, 2022-10-19:
I am hearing of a few in production, announced, etc vehicles that will have the ability to output AC power via the J1772 port. I would like to start collecting ideas, problems, standards, etc to plan for this feature via OpenEVSE. I know there is a lot of other dev work that is being done and needs done with higher priority, but "planning out loud" on this will help a more informed go / wait / no go decision about this feature.

List of vehicles that have AC output via J1772 port (work in progress):
- Lucid Air, Gravity, and future vehicles using their "wunderbox" inverter , 19.2kW 
 80 amps bidirectional AC , https://www.lucidinsider.com/2022/07/12/lucid-motors-tech-talk-on-wunderbox-electric-vehicle-charging-system/
- Polestar 3 and above (unconfirmed), https://emobilcar.com/news/polestar-develops-bi-directional-charging-technology
- Volvo EX90 (unconfirmed)
- Possibly VWAG (unconfirmed)

Relevant standards (work in progress / research required):
- ISO 15118 ?
- Other SAE equivalent
- Other IEEE equivalent

Possible Problems (work in progress):
- Signalling vehicle to start AC power export
- Signalling vehicle correct amps limit for circuit
- Receive start trigger via "two wire start" capable hybrid inverters? Potential solution: raspberry pi to watch for this signal and spit out HTTP API command?
- Signalling vehicle to stop export, resume import operations
- Signalling vehicle to stop export before handle release / removal
- Are the CTs / OpenEVSE board physically capable of reading / measuring amps in the other direction?

What have I missed on the above lists? Quick and dirty estimate of feasibility, difficulty to implement, etc?

KipK, 2022-10-20:
I have made my own V2L adaptor for my Ioniq5.
Important things to know is the com with vehicle to activate V2L is not standardized. 
There's different resistor values to use depending of the brand. ( Ex Kia/Hyundai != MG )

glynhudson, 2022-10-20:
@fhteagle V2L cannot be used for V2H/V2G since V2L cannot sync with the grid, V2L can only be used in island mode. CCS V2X (ISO 15118-20) as far as I'm aware requires additional communication with the vehicle which currently the OpenEVSE is not cable of. 

@KipK that's cool, is this documented somewhere? It's surprising they all use different resistors?! This is going to cause a lot of confusion when people try and use a Kia adaptor on a Hyundai!

KipK, 2022-10-20:
@glynhudson  Kia works on Hyundai, but MG not.

I used this scheme for mine : 

[image]

resistor between PE & PP is 62 ohm for Kia/Hiundai, and 70 ohm for MG

I saw on ali express some V2L CCS plug with a switch for different brands

fhteagle, 2022-10-20:
@KipK  - Very interesting about making your own custom V2L adaptor. Do you have it up and working already? Agreed that the state of standardization for vehicle power export leaves much to be desired. 

@glynhudson - Agreed that each user will need to possibly plan for additional equipment upstream of the OpenEVSE to fully support bidirectional charging, and those equipment requirements will likely vary by off-grid vs on-grid, differing jurisdictions, etc. There are hybrid smart inverters that are capable of utilizing multiple unsynchronized AC inputs. My personal target use case calls for the Sol-Ark 15k to accomplish the grid-interactive and auto-islanding functions. This equipment includes a GEN input that does not need to be grid synchronized. 

As to "additional communication with the vehicle which currently the OpenEVSE is not capable of.", the point of my opening this issue was to serve as a feasibility study for how much effort is going to be required to add this. Part of that process is getting a decent definition of scope of work. I think trying to support every individual manufacturer's quirks, communication add-ons, resistor values, etc that do not conform to any open standards should be out of scope. If the standards are not stabilized enough yet, then the answer of the feasibility study is "wait".  I am perfectly fine with that answer, as long as the feasibility study at least identifies "wait until when".

So, I vote that downstream equipment adapters, communication boxes, toggle switches, resistors for various OEMs, etc are out of scope for OpenEVSE project. Upstream hardware equipment above the OpenEVSE is likewise out of scope. The only considerations we should have for upstream and downstream hardware and communication are a well defined, open standards based method of changing from charging the EV to exporting power from the EV modes and back. Does this make sense?

KipK, 2022-10-20:
@fhteagle yes for monthes, works as it should. The Ioniq 5 / EV6 can deliver 3.6kw from v2l. Not bad when you have a long grid failure to backup the fridge / Wifi / smart home server.

[image]
[image]

fhteagle, 2022-10-20:
@KipK - Awesome. AFAIK, our US cars are limited to 120V and a lower kW export. I have not researched if a US car can be tricked into exporting 3.7kW at 220-240V range.............

Also, my understanding of ISO 15118 is that it utilizes powerline communications on the J1772 / Type 2 pins. Yes, this is utilized for communication in the CCS (DC) based power export system (which I also think is out of scope for OpenEVSE for now... Until OpenCCS project is started someday anyway lol). But ISO 15118 communications protocol can also be utilized over AC-only connections as far as I know. I have not bought or read the spec document itself as of yet, but I may do so in the future. If anyone has it, or has read it and has better information than I do, I am all ears. 

Thanks

fhteagle, 2022-10-21:
Other minor issue I thought of, is that if there is no source of AC power, the OpenEVSE boards will be likewise de-powered. De-powered OpenEVSE cannot speak ISO 15118 to the vehicle to start the export. If there is AC power already on the circuit, then the vehicle power export will need to sync to it, which as @glynhudson mentioned I do not think most cars can do.

Possible solutions: 

- Add a small capacitor or battery to allow OpenEVSE to remain powered for a few minutes to accommodate the changeover from charging to export. Ford uses this idea for the intelligent backup power system, the "dark star battery" I think they call it. In the case of the OpenEVSE, a small rechargeable 18650 cell or two would probably do it though (voltage and amp-hours needed TBD). However, we have a separate issue of how to charge or jump start the OpenEVSE battery again if the first changeover period goes too long and discharges the battery? 
- Auxilliary 5VDC USB-C, or 12VDC barrel plug input to OpenEVSE? This is my preferred solution, but would need to be designed in such a way as to not affect the weatherproof status of the case.

If anyone has any other brilliant ideas for powering the unit during the changeover, I would be glad to hear them.

fhteagle, 2022-10-25:
Possibly relevant: #29

glynhudson, 2023-03-30:
This is an interesting discussion but not really relevant to this repo, I will close this issue. It would be best to discuss on a form e.g https://community.openenergymonitor.org/

fhteagle, 2023-09-30:
@glynhudson could you turn this into a discussion item now that discussions are on for this repo?

Also, parallel work from EVerest project:
https://github.com/EVerest/logfiles

Looks like more cars already will pass back AC or DC power through their charge ports than manufacturers have been loud and proud about yet.

fhteagle, 2024-02-10:
Adding additional notes about this with new developments:

SAE J3400 specifications for the "Tesla" aka "NACS" connector and communication supposedly done. 

Tesla Powershare is in real world testing at employee houses, customer houses soon. https://www.tesla.com/powershare  . Supposedly, the Tesla 3V (not a typo, its 3V not v3?!?) gateway is alternately powered during a grid outage by a low voltage DC tap that goes vehicle -> EVSE -> gateway , not sure across which pins though . I assume this a proprietary / nonstandard implementation that is not actually written into J1772 / J3400? 

Also, second request to convert this to a discussion please @glynhudson or @jeremypoulter