OpenEVSE/openevse_esp32_firmware #458: Allow charging control to 1A

andriuskr, 2022-10-27:
Now Tesla through its API alows to set minimum charging rate to 1A. Why not allow it to OpenEVSE. Would be very usefull for Solar only charging.
[image]

chris1howell, 2022-10-27:
The minimum value defined in the J1772 spec is 6A. It is not possible to
send a lower value with the pilot signal.

andriuskr, 2022-10-27:
> The minimum value defined in the J1772 spec is 6A. It is not possible to send a lower value with the pilot signal.
> […](#)
Is it the same for 3 phase Type-2 mennekes ?

glynhudson, 2022-10-27:
>Is it the same for 3 phase Type-2 mennekes ?

Yes

closed 2022-10-27 by glynhudson

glynhudson, 2022-10-27:
It's very cool that Tesla allows charging down to 1A, I wonder what the efficiency is at this low level? Probably quite poor, but I guess if it's coming from solar it doest matter too much. To control the current less than 6A we would need to interface directly with the Tesla API this would not work very well for solar PV divert since there is quite a delay between an API request and the change in current being applied.