OpenEVSE/openevse_esp32_firmware #460: ESP gateway issues with factory config over serial

glynhudson, 2022-10-27:
The PR which added serial config seems to be causing an issue with ESP gateway https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/443. I've tested on Rev G using the new rev G environment. V4.1.5 works fine on the ESP gateway using the new rev G environment. 

Could there be a serial port conflict?

```
OpenEVSE WiFI fe2c
Firmware: 4.1.5
Build date: Oct 27 2022 17:21:13
IDF version: v4.4.1-1-gb8050b365e
Free: 265712
[    76][E][vfs_api.cpp:104] open(): /littlefs/schedule.json does not exist, no permits for creation
[    77][E][esp32-hal-adc.c:180] __analogRead(): Pin 1 is not ADC pin!
AP IP Address: 192.168.4.1
E (10434) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10434) task_wdt:  - loopTask (CPU 1)
E (10434) task_wdt: Tasks currently running:
E (10434) task_wdt: CPU 0: IDLE
E (10434) task_wdt: CPU 1: IDLE
E (10434) task_wdt: Aborting.

abort() was called at PC 0x4012f4cc on core 0

Backtrace:0x400836e9:0x3ffbee1c |<-CORRUPTED
```

jeremypoulter, 2022-10-27:
@glynhudson can you try with the debug enabled? You can see the required changes in the `openevse_esp32-gateway-e_dev` env

glynhudson, 2022-10-27:
```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

OpenEVSE WiFI fe2c
Firmware: 4.1.5
Build date: Oct 27 2022 18:00:05
IDF version: v4.4.1-1-gb8050b365e
Free: 265764
Got 65535 � from EEPROM @ 0x0c00
No JSON config found
Got 65535 � from EEPROM @ 0x0000
No JSON config found
No JSON config found, trying v1 settings
Loading config
Got '' 255 == 128 @ 575:32
Got '' 255 == 128 @ 0:32
Got '' 255 == 128 @ 32:64
Got '' 255 == 128 @ 542:33
Got '' 255 == 128 @ 128:45
Got '' 255 == 128 @ 173:32
Got '' 255 == 128 @ 438:60
Got '' 255 == 128 @ 732:96
Got '' 255 == 128 @ 205:45
Got '' 255 == 128 @ 250:32
Got '' 255 == 128 @ 282:32
Got '' 255 == 128 @ 314:64
Got '' 255 == 128 @ 378:30
Got '' 255 == 128 @ 408:30
Got 'ffffff'  255 == 127 @ 828:4
Got '' 255 == 128 @ 498:15
Got '' 255 == 128 @ 513:15
Got '' 255 == 128 @ 528:10
Got 'ffffff'  255 == 127 @ 538:4
Got '' 255 == 128 @ 607:45
Got '' 255 == 128 @ 652:80
After config_load_settings: 263656
[   120][E][vfs_api.cpp:104] open(): /littlefs/schedule.json does not exist, no permits for creation
[   124][E][esp32-hal-adc.c:180] __analogRead(): Pin 1 is not ADC pin!
Starting AP
(WIFI_MODE_AP == (WiFi.getMode() & WIFI_MODE_AP)) = 0
(WIFI_MODE_STA == (WiFi.getMode() & WIFI_MODE_STA)) = 0
Got Network event ARDUINO_EVENT_WIFI_READY
Got Network event ARDUINO_EVENT_WIFI_AP_START
Set host name to openevse-fe2c
Got Network event ARDUINO_EVENT_WIFI_AP_STOP
Got Network event ARDUINO_EVENT_WIFI_AP_START
Set host name to openevse-fe2c
AP IP Address: 192.168.4.1
E (10576) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10576) task_wdt:  - loopTask (CPU 1)
E (10576) task_wdt: Tasks currently running:
E (10576) task_wdt: CPU 0: IDLE
E (10576) task_wdt: CPU 1: IDLE
E (10576) task_wdt: Aborting.

abort() was called at PC 0x401a9ee0 on core 0

Backtrace:0x400836e9:0x3ffbee1c |<-CORRUPTED

ELF file SHA256: 0000000000000000
```

jeremypoulter, 2022-11-05:
@glynhudson Do you have the ethernet connected?

PR #533 (2023-03-08): Re-work of the network manager
This is a rework of the code to manage connecting to the network, WiFi or Wired. This code better manages the state we are in so hopefully there are less issues and is also more agressive at staying connected to the WiFi access point.

Fixes #420, fixes #460
closed 2023-03-08 by glynhudson