OpenEVSE/openevse_esp32_firmware #466: Manual Override button says "Start" but should actually say "Pause"

Edison517, 2022-11-02:
This button usually toggles to say the opposite of the current state, so that you can start a paused session, or pause a current charging session. But when I first load the page, the button already says "Start", even though a charge session just started after plugging my car in.

Does this toggle button need updated or is it just me? I'm on 4.1.5.

jeremypoulter, 2022-11-02:
Thanks for the report, can you send the output of the `/claims`, `/claims/target` and `/status` endpoints? IE go to http://<IP of OpenEVSE>/status in the browser

jeremypoulter, 2022-11-02:
Also a screenshot would also be helpful

jeremypoulter, 2022-11-02:
Thanks, I will take a look

jeremypoulter, 2022-11-05:
I have been able to replicate this issue. I think this is just a display issue, the button operates as expected. The translation cond is forcing the default value to `Start`, any state updatewill fix the issue. Just having a look to see if I can fix this.

jeremypoulter, 2022-11-05:
@KipK I had to add `pause` to the English translations

davidwkerr, 2022-12-07:
I have just upgraded from 8266 WiFi to ESP32. Using 4.1.6. This Start/Pause button problem is driving me nuts on both my Zoe and my Tesla. So, just adding my voice to hoping it can be fixed in the next release.

jeremypoulter, 2022-12-07:
Oops will get the fix for this merged.

closed 2022-12-14 by jeremypoulter