OpenEVSE/openevse_esp32_firmware #467: /settime endpoint problem

KipK, 2022-11-04:
There's somlething wrong with the time handling.

While integrating the set time using /settime endpoint to new UI, I've spotted a strange bug:

Example:
Current local time is 18h33

Sending to /settime:
ntp: false
tz: Europe/Paris|CET-1CEST,M3.5.0,M10.5.0/3
time: 2022-03-11T17:33:00.000Z  ( sending zulu time ISO format ) 

I receive fiirst from websocket:

{"time":"2022-11-03T**17:33**:26Z","offset":"+0100"}

then few seconds later receiving from /status:

time: "2022-11-03T**16:33**:27Z"

I can see the same problem with current UI when setting manually.

Same when setting with NTP, receiving first a time with one hour shift, then few sec later the time is back to ok.

But sending correct time manually ( with old UI too ) will end with a 2 hours shift

fhteagle, 2022-11-04:
I can confirm that issue has existed on the 4.x UI for quite some time. It gets positively maddening with UTC offsets in the -6 range. 

For example, manual time setting will even arbitrarily advance the day number to 1 more than what was entered in the GUI.

KipK, 2022-11-04:
Yes got this one with date too.

KipK, 2022-11-14:
> the acknowledgement is just that it has saved the time, not that it has retrieved it from NTP.

mhh , but in the last case, /settime only sent the ntp: true, and no time, so the time sent back should be the one retrieved by NTP isn"'t it ?

jeremypoulter, 2022-11-14:
Hum, will have to check..

KipK, 2023-01-17:
I have fixed this issue on UI2 side by setting /config time_zone to UTC when manual or browser time is set.

commit 29711656 (2023-03-13): Fixes to give the same time from the `/status` and the websocket

previously one was returning local and the other GMT. Now they both return GMT

Fixes #467
closed 2023-03-29 by glynhudson