OpenEVSE/openevse_esp32_firmware #468: Websocket only reports back pilot not actual max current

soloam, 2022-11-05:
The websocket only reports back the pilot value, never the actual max current value. According to the RAPI, we can get the "pilotamps" and the "cmaxamps". The pilot is the actual value, not saved on the EEPROM, used for by the shaper, for example. The  cmaxamps is the actual value selected by the user, and saved on the EEPROM. The cmaxamps is never reported on the websocket.

Thank You

soloam, 2022-11-05:
From what I see, that information is absent from all interfaces, websocket, mqtt and http request to status. The only way to get this value is to make a rapi request. This should be reported! It's relevant to use on other systems, for example Home Assistant.

jeremypoulter, 2022-11-05:
This is available on the `/config` endpoint as `max_current_soft` or `max_current_hard` depending on if you want the max value configured by the user or max value that the hardware can support.

soloam, 2022-11-05:
Yes, I found it! But I think it should also be in the web socket and the mqtt, so that it allows to sync values between multiple platforms.

Thank you

jeremypoulter, 2022-11-05:
For config values there is the `config_version` in the web socket that you can use to detect changes to the config and reload the config, you should see this if you load up multiple copies of the UI, change the current in one and it will be reflected in the other.

firstof9, 2022-11-05:
I am trying to avoid polling the devices as much as possible, would it be possible to potentially get the `max_current_soft` reported via web socket as well?

jeremypoulter, 2022-11-05:
@firstof9 I am trying to keep the websocket as a mirror of - `/status`, at some point I may switch to one of the various pub/sub protocols but that is not going to happen any time soon so that is why I added the various `xxxx_version` properties, including `config_version` just wait for a change of that value then fetch `/config`, no polling needed

firstof9, 2022-11-05:
Fetching `/config` is a poll technically, I'm requesting the data rather than it being push by the device.

I'll see what I can come up with, thanks!

jeremypoulter, 2022-11-05:
It is a compromise, short of rewriting that whole section as the current WS implementation isn't really suitable ;-)

firstof9, 2022-11-05:
I think I'll check if `shaper == 1` to not update the value in Home Assistant then once `shaper == 0` resume monitoring `pilot`.

jeremypoulter, 2022-11-05:
That seems like odd logic! What is the issue with what I have suggested other than it not being ideal? Works well for the UI.

firstof9, 2022-11-05:
I implemented this before the `config_version` thing was in place, to use the pilot to monitor the Max Current setting.

I'll keep tooling around to see what I can come up with. Either way I have more stuff to add to the library to parse. :+1:

soloam, 2022-11-05:
> I think I'll check if `shaper == 1` to not update the value in Home Assistant then once `shaper == 0` resume monitoring `pilot`.

This does not seams that it would work. The shaper isn't the only thing that suffers with this! For example the eco mode also would have this problem. Then the fact that the snapper is on doesn't invalidate changing the max current, I could go to the ui and change the max current to other value, or in a completely different interface, and that shou be reflected by the api!

firstof9, 2022-11-06:
I have it now using `max_current_soft` to set the select, and it fails back to `pilot` for older firmware. I have the `max_current_soft` get updated from `/config` when a `config_version` is thrown on the web socket.

soloam, 2022-11-06:
That's great! I'll test it out when available in beta

KipK, 2022-11-14:
you have max_current_soft in the config, but it can also be overriden ( on the lower side ) by a claim. So you should also check the 
/claims/target endpoint to see if a service has claim a max_current property.

KipK, 2023-03-10:
just for info, you also have schedule_version, limit_version, claims_version, override_version, plan_version , etc. each endpoint other than status has a version published over websocket to know when you have to grab new data for each endpoint.

firstof9, 2023-03-10:
> just for info, you also have schedule_version, limit_version, claims_version, override_version, plan_version , etc. each endpoint other than status has a version published over websocket to know when you have to grab new data for each endpoint.

Is this in the upcoming 4.1.8 version? I don't see these in 4.1.7.

KipK, 2023-03-10:
mhh, perhaps not all of them ( limit is still in the dev builds  ) , in /status you should be able to see them

    "config_version": 0,
    "claims_version": 43,
    "override_version": 0,
    "schedule_version": 2,
    "schedule_plan_version": 3,
    "limit_version": 2,

I've added some while doing the UI so yes the missing ones should be for upcoming release

firstof9, 2023-03-10:
I will update my library to account for these as well.

closed 2023-03-30 by glynhudson