OpenEVSE/openevse_esp32_firmware #472: Enhancement: toggle three/one phase for Grid (+I/-E) MQTT Solar divert

jippej, 2022-11-10:
First of all great work on openEVSE and all the work you guys do.

In Short:
I would like to see implemented a toggle switch in the app that is setting one or three phase value for the Grid (+I/-E) MQTT topic.
Division for three phase will then be done in ESP32 instead of at "customers" side

Background, as discussed in the forum topic: https://openev.freshdesk.com/support/discussions/topics/6000063453

jeremypoulter, 2022-11-10:
Hi, sorry, I am not sure I understand what would be different between the two modes, can you clarify?

jippej, 2022-11-10:
It is the three phase issue from europe, if my meter (three phase) is telling I am feeding back 1,4kw, this will be around 3,4 amp per phase. However openEVSE will start charging with 6A (on each phase) and basically will pull around 1kW from the grid. So basically the firmware will need to divide the kw value by three in case of a three phase meter.
Basically the firmware will need to know if it has to divide by 3 or not. 
Additional a toggle switch will also very handy in case your three phase charger is sometimes charging three phase capable cars and sometimes one phase only capable cars (I will not go into detail on the possible phase imbalances here). it will make the switch over easy to do via the app.

sorentorp, 2022-11-24:
agree on this one

KjellVerb, 2023-01-21:
Thirded, would be very nice!

jeremypoulter, 2023-01-27:
https://community.openenergymonitor.org/t/correct-power-calculation-for-booth-mono-and-triphased/22587

PR #528 (2023-01-29): Adds is_threephase virtualbool switch.
 Factors _monitor.getAmps() by 3 in _evse.getAmps() if true

This should fix current_shaper & divert mode beeing wrong with threephase, and displays on UI.

fix #472

still needs evse module fw compiled with -D THREEPHASE
closed 2023-01-29 by glynhudson