OpenEVSE/openevse_esp32_firmware #479: Restarting controller via HTTP throws error 500

ruimarinho, 2022-11-15:
Hi,

I'm on 4.1.6-dev and attempting to restart the controller via HTTP (using the UI or simply doing a curl to `https://openevse/r?json=1&rapi=$FR`) results in an error 500 with response `{"cmd":"$FR","error":"RAPI_RESPONSE_TIMEOUT"}`.

Interestingly the HTTP request is working because looking at the controller's terminal I see `$FR^30` followed by the asynchronous boot command `$AB`.

glynhudson, 2023-03-31:
I'm able to replicate this issue, but it does not seem to be caused a problem since the controller still restarts. But I agree it would be good to not have the error. 

[image]

KipK, 2023-05-16:
This will be solved by PR #595 commit https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/618/commits/c1d93a1b42739797592a8dacacdddee9f1b44aca

PR #618 (2023-05-21): New settings / flags (#595)
FIxes #595
needs UI branch "newflags"

Not to be merged before UI migration

done: 
 * wizard_passed ( fw + ui )
 * divert_type ( fw + ui ) 
 * vehicle_data_src ( fw + ui )
 * default_state (fw + ui )
 * set AP ssid & password ( fw + ui ) 
 * added /restart {"device": "gateway|evse"} to http api & mqtt + restartEvse() method in evseManager ( close #479 ,
closed 2023-05-21 by jeremypoulter