OpenEVSE/openevse_esp32_firmware #480: Random safety counters and disabled checks after restarting controller

ruimarinho, 2022-11-15:
Hi,

My car has a very tricky charging software which requires a controller restart to be waken up. Even the sleep/disable toggle added on a (somewhat) recent firmware didn't solve the problem for me. The only thing that does is restarting the controller.

To that end, every charging sessions starts with a controller restart via `$FR` RAPI command. That used to work out well but most recently (4.1.5 and 4.1.6), safety checks are randomly disabled and counters sometimes all show up as `-9568`. 

Obviously this is a safety concern for which I don't have a good explanation for except recent WiFi firmware upgrades.

[image]

In this screenshot, all values are zero, but I've just restarted the controller again and now one of them shows as 18 with two safety checks disabled.

[image]

glynhudson, 2023-03-31:
It's unlikely that the safety checks are actually being disabled when the controller is restarted. More likely a bug in the way the UI is reading the status of the checks. Could you check to see if this issue is present in GUI V2?

KipK, 2023-05-16:
@ruimarinho , what vehicule brand/model do you have that needs an evse restart ? We're working on this issue for now and needs some data.

ruimarinho, 2023-05-16:
@KipK Volvo XC60 T8 (PHEV)

KipK, 2023-05-16:
Thanks, can you try something for me ? 

Let the car plugued as usual, and evse in sleep mode.
Try to wake it up by sending ( from Developper tab ) those rapi commands:
$FD  ( this will set the evse to disabled instead of sleep ) 
wait few seconds, then send $FS ( this means activate evse ) 

I'd like to know if the car has been waken up and charge is starting then

ruimarinho, 2023-05-16:
@KipK with any specific WiFI or controller firmware?

KipK, 2023-05-16:
Yes, either V1/v2 UI you can send rapi commands in developper section

KipK, 2023-05-16:
You'll have to enable the charge too on UI or the wifi fw will try to switch it back to disabled probably.

ruimarinho, 2023-07-07:
@KipK it looks like sending `$FD` was enough to wake the car.

Initially, a manual override would only lead to this state like this:

[image]

After sending `$FD`, the car started charging. Curiously, I am almost 100% sure that I did not have to issue the `$FS` command - could there be some other check automatically re-enabling it?

Thanks!

closed 2026-09-21 by github-actions[bot]