OpenEVSE/openevse_esp32_firmware #481: Controller restart ignored eco divert mode setting at boot

ruimarinho, 2022-11-15:
If OpenEVSE is in Eco (PV Divert) mode and the controller is restarted, those settings are ignored (but are successfully restored) and the car starts charging right away:

**Before restart** 

[image]

**After restart**

[image]

As you can see, Eco mode is still active after restarting the controller, but the boot logic ignored it was active and there was no energy being exported at all.

KipK, 2022-11-16:
this is default behavior, if you want to disable Divert that survive reboot, it's on the Config / Services part.

ruimarinho, 2022-11-16:
Not sure I follow -- in order to enable this mode, the service needs to be enabled, but that hasn't changed, has it?

[image]

KipK, 2022-11-16:
ho I have missread sorry this effectively shouldnb't happen

KipK, 2022-12-11:
Prolly related to #452 and #450

Seems EVSE is always starting to max amp whatever the current claim is. Sending another claim after start get back to normal behavior.

PR #532 (2023-02-06): fix #452 reevaluate claims if amps > pilot
Until the bug is corrected on evse fw, this solves the issue and prevent the charge_current going to max when Evse goes from Disabled to Active state ignoring claims & pilot values
fix #452 and should also fix #513 fix #388 fix #481

Edit: It seems those bugs are related to PP_AUTO_AMPACITY beeing set. Posted an issue there: https://github.com/lincomatic/open_evse/issues/156?fbclid=IwAR37MVzzv2WZ7iDPPr3lkryBSUIvsDBsbPJkFu5PcZG1JgMT7e2LF6KxFzQ

So this is not really an OpenEvse bug, but more that OpenEvse and Wifi module don't share the non tethered ch [... cut]
closed 2023-02-06 by jeremypoulter