OpenEVSE/openevse_esp32_firmware #484: Feature Request : Daily Energy Counter

TommySharpNZ, 2022-11-17:
We have a session and total energy counter in kWh....
I'd really appreciate if we could also have a "daily" counter that resets daily at midnight.

TommySharpNZ, 2023-01-28:
Any further thoughts on getting a daily counter added similar to the session and total counters?

firstof9, 2023-01-29:
I could go for a session counter that counts up while charging.

TommySharpNZ, 2023-02-15:
Based on me trying to chart the Total Usage am I correct in deducing that the counters are only updated at the end of each charging session?
With so many "time of use" tarrifs now it would be more ideal to update the counters every 10 seconds or something like that...
[image]

PR #556 (2023-03-11): replace old counter from OpenEVSE module with new EnergyMeter class on ESP
Depends new network branch to be migrated first.
now gives :
- session_elapsed
- session_energy 
- total_energy
- total_day
- total_week
- total_month
- total_year
- total_switches ( number of time the relay/contactor has turned on/off )

Old OpenEvse total is imported to new counter when migrating first time, then a flag prevent to import it again if counters are reset.

fix #484
closed 2023-03-11 by jeremypoulter