OpenEVSE/openevse_esp32_firmware #491: Time format in Timer

TommySharpNZ, 2022-12-01:
I have setup a timer for 9pm to 6am. But when I look at the timer now it's not clear if it's 9am or 9pm? Maybe this should be 24 hour format?
[image]

TommySharpNZ, 2022-12-01:
In the new UI currently being worked in looks like this may be resolved :-)
Although I'm not sure when that will be available for general release.

[image]

davethetallguy, 2022-12-01:
What version? Mine (version 4.1.5) pretty clearly has AM/PM indication. Does this infer without the indication that it is in 24 hour mode?  
[image]

TommySharpNZ, 2022-12-01:
When I am setting up the Timer it does show am/pm....
Bat after that is toes not...
Could be Firefox issue maybe?

jeremypoulter, 2022-12-01:
Yeah, we use the browser's input for the time as these should give the appropriate experience for the platform

TommySharpNZ, 2022-12-01:
Confirmed it works fine in the Edge browser.

jeremypoulter, 2022-12-02:
Just tried Firefox and at least with my settings, it displays in 24 hour:
[image]

But equally, so does Chrome:
[image]

KipK, 2023-01-05:
new UI use now locale settings to display time in 24H or 12 am/pm format.

closed 2023-03-30 by glynhudson