OpenEVSE/openevse_esp32_firmware #492: Eco (PV divert) mode keeps turning off

TommySharpNZ, 2022-12-01:
My goal is to us Eco mode all the time except for 9pm to 6am which is when we get super cheap power.

I have setup a timer from 9pm to 6am but this always seem to turn the Eco mode off.

I feel I should actually be setting 2 timers so that from 6am to 9pm the charger goes into eco mode... But I don't think this is possible?

I guess I'm just a bit confused here...

jeremypoulter, 2022-12-01:
Can you describe what you mean by the eco mode turning off? What you have setup should do what you are describing, as in eco mode 6 to 9 and full power 9 to 6

TommySharpNZ, 2022-12-04:
From what I can tell, the OpenEVSE is still in "eco" mode because it is charging the car at a variable rate as the "Current Now" value does seem to change.

KipK, 2022-12-11:
There's chance it's just the /status JSON_OBJECT_SIZE that needs a bit of increase like I did yesterday to fix the missing published properties in /config.
I've increased it also for status in this Pull Request, could be nice to test it for this issue too
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/497

KipK, 2022-12-11:
Can you try this build and tells if it fixes the problem ? 
This is for openevse_wifi_v1 module.
[firmware.zip](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/files/10202176/firmware.zip)

TommySharpNZ, 2022-12-11:
That seems to resolve the issue and the web interface now always shows the correct state :-)

PR #497 (2022-12-14): fix  #496 #492 increased json object size
fix #496 
fix #492
closed 2022-12-14 by jeremypoulter