OpenEVSE/openevse_esp32_firmware #505: Release 4.1.7 spams MQTT with non-stop evse/claim: {"state":"null"} messages many times per second

rigorka, 2022-12-30:
Steps to reproduce: 
- I had 4.0.0 installed, upgraded that via the web UI to 4.1.7.
- My MQTT handler has collapsed immediately overloaded by hundreds of evse/claim: {"state":"null"} messages.
- I've tried Factory Reset which did not fix the situation
- Making and then releasing a Claim via HTTP API fixes the problem until the next reboot
- After rebooting the Openevse Wifi it spams the MQTT queue again

Downgrading to release 4.1.6 fixes the problem.

glynhudson, 2023-01-01:
@KipK could this be related to https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/488 ?

KipK, 2023-01-02:
Mhhh I don't think so. I'm jumping in a plane right now but will get a look tomorrow. 
I haven't saw such behavior here,  but there's definitely something wrong here I have to track down.

KipK, 2023-01-02:
Ok I've found it's prolly there:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/65ca1bc4a877eb618a3c44be11019481dd75d11c/src/mqtt.cpp#L501

I'll fix that

KipK, 2023-01-03:
Forget it, I've found. I forgot to inialise claims_version ...
Fix is coming

closed 2023-01-05 by jeremypoulter

KipK, 2023-01-06:
It stills not fixed I can see the debug log with mqtt activated spaming claims has changed, posting claims. I'm looking at it.

commit 27d724a7 (2023-01-05): Merge pull request #507 from KipK/#505

fix #505 missing claims _version init
commit c568dd8e (2023-01-05): fix #505 missing claims _version init
commit 44f4c5bc (2023-01-06): Merge pull request #511 from KipK/#505

fix #505
commit 5cfa0209 (2023-01-07): fix #505 missing claims _version init