OpenEVSE/openevse_esp32_firmware #508: Current Shaper is not working as expected

tux43, 2023-01-05:
Hi,
I've setup up 'Current Shaper' as per below.

[image]

Unfortunately when I set the MQTT topic it isn't taking effect.

KipK, 2023-01-05:
You've set it wrong, the livepowerload topic shouldn't be in the path of Base Topic ( i e openevse_9c0c in your case ) , this is the topic where OpenEVSE publish data. 

The Live Power Load topic is not populated by openevse but your external smart home controller, so it should be on another base topic like homeass/energy/livepower

tux43, 2023-01-05:
> The Live Power Load topic is not populated by openevse but your external smart home controller or emoncms, so it should be on another base topic like homeass/energy/livepower

Does it take the mqtt server from the enabled one (right hand pane?). The way the page displays it looks independent without a mqtt server specified. I’ll give this a shot in the morning when the sun is up!

Thanks

KipK, 2023-01-05:
Yes it does indeed.

tux43, 2023-01-06:
Unfortunately I am still having trouble with "Current Shaper". I have defined a topic within MQTT "homeenergy/energy/power" and set a nominal value of 1200 (watts).

[image]

I have setup OpenEVSE as per this screen with the intention of reducing the watts to the "Current Shaper"

[image]

It still charges at the full rate.

[image]

What am I missing?

tux43, 2023-01-07:
> The live power data needs te be refreshed each 360sec max or shaper will Pause EVSE .
> Posting fake data once won't work.

Ok, great. Although I am still having major issues trying to get it to operate effectively. It seems to be reading all the MQTT topics however it does not start charging with a "Waiting for Solar message as per below"

Here is what I have done.

I have defined the following MQTT topics all in watts. These are updated every 30 seconds.

homeenergy/energy/availablepower (Feed in or consumption from the grid - can be positive or negative)
homeenergy/energy/totalsolaroutput (Solar output and will always be positive)
homeenergy/energy/housepowerload (House consumption this and always negative)

[image]

I have then defined MQTT as per below and consumed a voltage topic so I get an acurate reading.

[image]

I have defined Solar PV Divert to read from the Total Solar output as per below.

[image]

I have defined the current shaper to montior the household load, I assume this is subtracted from total solar output to determine the current to send to the vehicle?

[image]

After doing all of this OpenEVSE is reading the values however keeps stating it is "Waiting for solar"? 

[image]

KipK, 2023-01-07:
Shaper is disabled when Divert is active*Your problem now concern Divert then, out of topic so.

btw, your solar input is 811W , not enough to start a charge as minimum is 6A, ie 1428w for 220w so it's totally normal nothing happends here.

closed 2023-02-17 by tux43