OpenEVSE/openevse_esp32_firmware #509: /evse/console websocket issue

KipK, 2023-01-05:
Websocket with evse console close with thrown error:
` Could not decode a text frame as UTF-8. `

Seems /evse/console endpoint should filter its output first

jeremypoulter, 2023-01-06:
Which browser/client is that with? I remember seeing something like this with a Node JS client but think it worked ok with most browsers

KipK, 2023-01-06:
All legacy browsers, it's mandatory to reject non UTF8 message now over web socket 
It's also the case on old UI , you just have to throw the web console.
It stills update on screen because it fetches from HTTP get at each reconnection.

jeremypoulter, 2023-01-06:
:( ok will take a look

jeremypoulter, 2023-01-06:
Guess I need to do the reverse of this: https://github.com/jeremypoulter/OpenEVSE_Lib/blob/master/src/openevse.cpp#L831-L836 won't be a true representive of what is being sent on the serial... 

Another option may be to send as binary and handle any processing in the browser, may be a bit easier with more memory/processing

KipK, 2023-01-07:
Why not just finding a magic char in the utf8 table?

jeremypoulter, 2023-01-07:
Would have to copy all the data as utf-8 will be larger than the raw data. I will have a look at it again and see how plausible this is, but would like to keep this fairly light weight on the WiFi module side...

KipK, 2023-01-07:
I can try to use binary over ws if you can post a patch with binary output on this endpoint to test with.

KipK, 2023-01-07:
Or we just poll the http endpoint, without ws ( not really clean though )

PR #515 (2023-01-25): Quick and dirty filter to ensure no >127 chars are sent over the websocket and hense valid UTF-8 is sent
On a quick scan there does not look to be a readily avalible library for ASCII to UTF-8 conversion, so as we don't have all that many cases where this is an issue, basically just the space on the LCD, just a quick hack to ensure valid UTF-8 is sent.

Fixes #509
closed 2023-01-25 by glynhudson