OpenEVSE/openevse_esp32_firmware #512: Add a way to set mode 254 or 255 from the API

KipK, 2023-01-09:
Some Zoe users needs to setup a special procedure to prevent the Zoe going in sleep state .
The /config set Sleep/Disable state for pause has been introduced for that.
But it doesn't seems to solve the issue of all Zoe.
They have a  start procedure like this to solve their issue;
- stop the car
- before it goes to sleep mode, plug the car and start a 30sec charge.
- set the EVSE to sleep ( 254 vi $FD command)
- when low energy cost hours start, put the EVSE to Disable ( 255 via $FD command) 
- wait 30sec,
- Start charge

And then it seems to works.
The problem here is they still use RAPI commands for that, could be usefull to set an interface over the API to let them control that disabled state.

The /config one needs ESP32  to be restarted to let the disable state change effectively.

KipK, 2023-01-09:
Yep  have a way to dynamically set over the api the disable state from 254 to 255 or vis versa without having to reboot EVSE & write to the flash nor using RAPI commands.

mathieucarbou, 2023-05-14:
FYI I have a Zoe ZE50 E-Tech R110, and this vehicle is subject to the sleep issue.

Ref: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/223
Ref: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/643

So I added an automation in HA which is sending through mqtt a sequence of `$FD $FE` when vehicle connects. I don't need to start a charge.

```yaml
automation:
  # 
  - alias: "OpenEVSE: Zoé Bugfix"
    trigger:
      - platform: state
        entity_id:
          - binary_sensor.openevse_vehicle_connected
        from: "off"
        to: "on"
    condition: []
    action:
      - service: mqtt.publish
        data:
          topic: "openevse/rapi/in/$FD"
      - delay:
          hours: 0
          minutes: 0
          seconds: 2
          milliseconds: 0
      - service: mqtt.publish
        data:
          topic: "openevse/rapi/in/$FE"
```

So far, and like reported in the issue in reference, this is pretty reliable.

I've seen this option in OpenEVSE:

[image]

But I don't want to use it because of this warning:

[image]

I am using EVSE with an HA integration. **I need and rely on the EVSE to be able to detect when a car is plugged.**, which is translated in mqtt with `vehicle = 0 or 1`.

[image]

Considering RAPI through mqtt will be removed, I think we definitely need some mqtt topics to enable/disable EVSE.

I recently introduced `/restart` to restart the gateway.

Could we introduce something like  `/evse_enabled` and `/evse_enabled/set` set to `0` or `1` to be able to control that through mqtt ?

Thanks!

KipK, 2023-05-15:
It does but once saved in the config, and rebooted the evse.  not really usable for scripting some anti-sleep pattern yet. 
Perhaps a new parameter on the claim engine to set the disable mode when disabling evse state?

jeremypoulter, 2023-05-15:
No, I don't think the higher levels should be making those kinds of decisions, they should just start/stop and the OpenEVSE should deal with the interactions with the vehicle.

We need to define more sleep modes and config around how the charge is paused and resumed.  The thing that is not clear to me is the sequence of events that will put a particular vehicle to sleep or wake it up

KipK, 2023-05-15:
depending of Zoe BMS software version, they don't have the same procedure or bugs so this will be quiet complicate to find something universal.

jeremypoulter, 2023-05-15:
Doesn't have to be universal as such, but needs to have the flexibility to configure different patterns. I think we might need a mode to sleep, wait for n seconds then disable. Then also a module that forces a (min power?) charge for n seconds after the vehicle is plugged in. I think that should cover most of the cases I remember seeing. But I would like to see a list of vehicles and the associated hacks

KipK, 2023-05-15:
@mathieucarbou , don't you need the first procedure step describe at top of the thread on your Zoe? ( Trigger a charge few seconds before set the Zoe in sleep mode )

KipK, 2023-05-15:
On UI side we could have 2 selector menus for pause mode and activating charge:

Pause / Disable / 30s charge then Pause 

Charge Activation:
Activate ( default ) , Disable then Activate

But I'm not fan of this solution as it's a hack for buggy or non standard vehicles, keeping them at the same level deserve main clarity. 

Also I'm not even sure the actual Pause / Disable switch is used by somebody and solve any issue. There's no report on the  related Issue supposed to be solved by this commit so perhaps there's no pro at all to always keep the vehicle in disabled state when paused. This would be interesting to have some feedback.

If this Pause / Disable it not enough and those reported this got the same Zoe / Dacia problem, then we only need to match the top procedure and would simplify the setup:

Zoe / Dacia sleep & wake-up hack:
Enable / Disable  ( + description in help icon )

mathieucarbou, 2023-05-15:
> @mathieucarbou , don't you need the first procedure step describe at top of the thread on your Zoe? ( Trigger a charge few seconds before set the Zoe in sleep mode )

No! 100% sure. I didn't know about these steps, the only thing I need to do is disable then enable evse once the vehicle connects. 

I am always using Solar divert. So I usually plug when there is not enough sun, so no charge, and the vehicle wakes up at one point and charges when asked to.

KipK, 2023-05-15:
@mathieucarbou on what they told me, the vehicle can't wake up only after few hours. Guys using this procedure mostly plug their vehicle when going back from work, and startautomatically the charge in late night few hours later. 
Perhaps on your test there was not enough waiting time to see this behavior ? 

but I've also heard that some up to date BMS on zoe doesn't suffer from this.

mathieucarbou, 2023-05-15:
> @mathieucarbou on what they told me, the vehicle can't wake up only after few hours. Guys using this procedure mostly plug their vehicle when going back from work, and startautomatically the charge in late night hours later. Perhaps on your test there was not enough waiting time to see this behavior ?
> 
> but I've also heard that some up to date BMS on zoe don't suffer from this.

Oh!!! Yes that is possible... So to test that, I will have to keep the plug connected in the evening through the night and keep it connected in the morning until the sun rises. So it will be a test for this week-end :-)

KipK, 2023-05-15:
I've also contacted one of those guy to see if they can test your procedure and get same result.

jeremypoulter, 2023-05-16:
Thanks to @glynhudson we found some interesting info on the Zoe:
https://myenergi.info/zappiv2-compatibility-mode-settings-t2521.html
https://www.asefa-cert.com/medias/ev-ready-certification-technical-reference-document_1.4g1_20112018-non-signe.pdf

mathieucarbou, 2023-05-16:
> Basically, the Zoé stops the charge if the signal Zappi is sending to the Zoé is lower than these values. This means that the minimum charge rate is higher.
Zappi checks for "reactive current", as mentioned in EV Ready and adjust the PWM (the signal) according to the values in the Compatibilty Mode. 
This ensures that a charge can happen regardless of the EV being plugged in.

FYI, it does not apply to the new Zoe ZE50 (52kWh) E-Tech R110 because I am charging with solar divert during the day at a pilot of 6A - 10A depending. And I was able to confirm that with an OBD device and CanZe. The vehicle charges at a lowest current of 5.6A (pilot set to 6A) up to 19A when the pilot is set to 20A.

Also, at a lower current, the PF is about 0.8. But at higher current, the PFC triggers and the PF is higher than 0.95, most of the time 0.98 (measures with a shelly em dedicated for the EVSE box).

glynhudson, 2023-05-16:
I can also confirm the ZE40 'R' model can handle 6A charging. I think the 7.2A minimum limit that's mentioned in the links posted only applies to 'Q' model Zoes.

The Renault EV ready document has some info about the wakeup procedure, I'm guessing that 'non-compliant' modes is referring to older Zoes?! 

[image]
[image]

KipK, 2023-05-17:
The other way would be to use Renault API to send a start_charge when openevse switch to active. This will wake up the vehicle:

https://renault-api.readthedocs.io/en/latest/index.html

mathieucarbou, 2023-05-17:
> The other way would be to use Renault API to send a start_charge when openevse switch to active. This will wake up the vehicle:
> 
> https://renault-api.readthedocs.io/en/latest/index.html

No... Not an option. That's the API I was using before, **and that is the reason I bought the OpenEVSE box.**

- The Renault API is not stable and reverse-engineered
- Not supported by Renault
- Their servers are often down, causing the API to fail
- There is a rate limit preventing doing too much requests within 7 min
- They are changing their API keys sometimes, and tthe devs have to spy on the request / response from the updated mobile apps to their server to grab the new api key.

So no, please no... 😆

mathieucarbou, 2023-05-20:
OK so yesterday I've remove any automation I got. So no more $FD-$FE, or $FR.
Also, I don't use the disable sleep option.
Zoe ZE50 (52kWh) E-tech R110.

And today the solar divert started the charge normally when possible.

[image]
[image]

So I'm baffled.... 
Because I am 100% sure I ran into a sort of sleep issue one time, and I had to wake up the car to be able to start the charge.

So 2 explanations:

1) my spouse woke up the car unexpectedly by walking next to it (which unlocks it) 😁 
2) or the Zoe I have is not subject the issue, and the problem I saw was caused by something else

So I will keep testing, all automations disabled, sleep mode active.

[image]

jeremypoulter, 2023-05-21:
I may be able to get access to a Zoe to test this issue, can someone confirm the time it takes for a Zoe to go to sleep after the EVSE goes to sleep (just to reduce the testing time)?

jeremypoulter, 2023-05-21:
On actually 'fixing' this issue. I will add a simple module that adds a high priority claim to vehicle connection and an option to disable before waking up, with 4 config options
- charge_vehicle_connection, enable/disable the charge after vehicle connect, default false
- charge_vehicle_connection_duration, the duration of the charge, default 30 seconds
- charge_vehicle_connection_current, the current, default 8A (I assume this needs to be low to avoid exsessive import at a possibly high rate)
- disable_before_active: disable the EVSE module before waking when switching to active mode

mathieucarbou, 2023-05-21:
> I may be able to get access to a Zoe to test this issue, can someone confirm the time it takes for a Zoe to go to sleep after the EVSE goes to sleep (just to reduce the testing time)?

FYI I left the vehicle connected the night of sat-sun and the vehicle started charging sunday morning with solar divert (export mode) without any issue. Nobody woke up the car.

So I just think I ran into a random issue which had the same symptoms of a car sleep (because I had to woke up the car to start charging).

But from my testing during these 2 days, the Zoe ZE50 E-Tech R110 is not subject to the sleep issue... I will continue to monitor: I usually leave the car plugged during the night.

mathieucarbou, 2023-05-21:
> * charge_vehicle_connection_current, the current, default 8A (I assume this needs to be low to avoid exsessive import at a possibly high rate)

Why is the default not equal to the min_current ?

jeremypoulter, 2023-05-22:
> > * charge_vehicle_connection_current, the current, default 8A (I assume this needs to be low to avoid exsessive import at a possibly high rate)
> 
> Why is the default not equal to the min_current ?

It is my understanding that some Zoes need a min of 7.2 amps to start a charge, so thought it made sense to be cautious

jeremypoulter, 2023-07-05:
Can we try and colate information on which vehicles suffer from this issue? and if there are just (Zoe) firmware updates that are needed to be applied.

mathieucarbou, 2023-07-05:
> Can we try and colate information on which vehicles suffer from this issue? and if there are just (Zoe) firmware updates that are needed to be applied.

Hello, FYI I have disabled the automation since our previous discussion, and I didn't activate the special mode in OVSE. So I don't restart the box anytime and I have pause status set to sleep (defaults).

[image]

And everything works fine! We are charging and using the VE every day and on week-end or mornings it starts with divert when it has to. 

Zoe E-Tech R110 52kWh.

bubupol, 2024-01-15:
Hello I have a Zoe R90 Z.E.40  from May-2018. 
I have the same issues. It has been working well for few months with the setting to sleep, and since last week the car doesn't wake up.
I am baffled too. 
I tried both setting of the Pause Status : disable or Sleep. no success.

I can wake up the car with the remote by pressing the "lock doors" button and the charge starts.

1. I use home assistant, I was wondering to clone the remote radio signal to trigger the car to wake up ? 
2. Has some people managed to do some firmware updates to solve the issue ? (I will contact Renault).

.

MilorES, 2024-03-06:
OpenEVSE | 8.2.0.EU
OpenEVSE Wifi | v5.1.0_modified

I have a Dacia Spring from September 2022.
I usually plug it in before entering the house.
In Spain sometimes the economic time starts at 00:00 and we can also have the highest power at a low cost.
For the charging to start, I must go to the application or restart the ARDUINO, I seem to remember that restarting the ESP32 does not solve the problem.

When you restart the ARDUINO it returns to AUTO mode even if it is ON. Is this behavior normal?

Now that RAPI is not working, what commands are sent to restart or force it to start (it was not clear to me).

Note: I hope you can understand because I am using a translator since my native language is Spanish.

P.S. Having to wait until 00:00 instead of being able to rest before 23:00 makes me sad.

NEVdataDyne, 2024-12-03:
### **Dacia Spring 2024**
This is my best solution so far to allow home assistant to control the charge through openevse without:
- The car ever going to deep sleep
- Setting the pause status to disable (which doesn't work well on this car anyway)
- Using MQTT
- Using extra hardware to disconnect the data line between openevse and the car

### Solution (everything done in HA)
**Part 1**
1. When the car is being plugged-in, reboot openevse
2. Wait three seconds
3. Perform an override on openevse with status active (action : OpenEVSE: Set override; I put 6A, minimum current)
4. Wait 10 seconds minimum (maybe even less works; openevse status changes to charging for at least a couple seconds)
5. Then either charge or perform an override on openevse with status disabled (sleep)

**Part 2**
1. Every hour, if the car is not charging, perform the same action as when the car is being plugged in (didn't try for an even lower frequency, maybe every 2 hours works)

I tried to use the "pause status" set to "disable" in openevse but it gave me wierd results, sometimes the car would continue to sleep for some time then wake up and start charging, sometimes it would never wake up.

There seems to be at least two level of sleep in the Dacia Spring 2024, rebooting openevse to wake up the car by reseting the data line between openevse and the car seems to be working when the car is not sleeping too deep but it doesn't work if the car is sleeping for too long, that's why by reseting openevse and charging a couple of seconds every hours the car is not allowed to go in deep sleep mode.

kwauchope, 2025-01-25:
I've had this issue as well, it may not be palatable for everyone but posting my solution in case it suits someones needs.

I ended up patching the open_evse firmware to toggle a pin when charge is being enabled. I then connected that pin to a relay and bought the PP-PE resistor from the plug into the EVSE and connected it to the NC port of the relay. The enabled pulse temporarily disconnects the PP-PE resistor, waking the car back up when its reconnected. The cable that came with the OpenEVSE kit had a spare PP wire unused so was relatively straight forward.

cons:

* requires a hardware modification to EVSE and cable
* only useful in tethered installs (and no adapters as they all seem to use their own PP-PE resistor)

pros:

* unlike the pause status, car connection status is known
* should work on all vehicles, no matter what their quirk is
* doesn't require a short charge

On pros as well, i'm not sure what the ramifications are of stopping a car going to deep sleep as in the above solution so it may be another one. Presumably it eats into the 12v battery, requiring it to be topped up more often and reducing its life?

[0001-add-ENABLED_PULSE-pin-support.txt](https://github.com/user-attachments/files/18544904/0001-add-ENABLED_PULSE-pin-support.txt)

bubupol, 2025-01-26:
Hello @kwauchope ,

Thank you for this.I used to have a podpoint and it didn't have any problems waking up the car. I alos noticed that  a reset, reboot of the openevse can wake up the car. 

Any chance to have more details. 
Which version of the openevse you have (V5.5, or which year you bought it?)
Do you have pictures of your hardware modification ?
the specs of the Additionnal relay ? What do you mean by PP wire ? (standard Polypropylene wire? )

I need to process your text file.

bubupol, 2025-02-08:
@kwauchope , thank you for your detailed answer.

you are right. First I will test a bit more with the pause setting (disable/ sleep) after a reboot. monitor the behaviour.

Today, I use home assistant Renault integration and I can trigger a "start charge". but that rely on the renault cloud servers to be online and not them to mess around with the APIs. (they have been reliable recently, but terrible  month ago)

MilorES, 2025-04-07:
[image]
What do you select?

NEVdataDyne, 2025-04-07:
I use "sleep".
Before I found that "Disable" didn't solve the issue so I prefered to use the method described above (charging for a short time when connecting the vehicle)

bubupol, 2025-08-21:
Hi @NEVdataDyne ,

I use it the control "Start Charge" in the home assistant renault integration for my Zoe 2018. 
so in my automation, the last action is to send a "start charge" to the car. 

I am not entirely happy with the the solution because it has to go through the Renault online servers and their availailability/reliability was terrible.
in the last 6 months, Renault must have done something right, it has been very good. and the start charge works everytime. it wakes up the car.
Low tech option : I was considering trying with a switchbot / fingerbot to press lock on the remote on the car to wake it up.
Other option to explore  : plug something on the odb II port ? like an ESP32, and send a signal to wake up the car ? i don't know if that would work.

closed 2026-09-21 by github-actions[bot]