OpenEVSE/openevse_esp32_firmware #519: charging over kwh limit

danpadure, 2023-01-18:
I noticed a few times that on the latest version (4.17), the kwh limit is not working

The limit was set to 5kWh.  
From the logs we can see that the limit was reached, stopped for a brief moment and then continued to charge until I stopped it at 9.25kW (I switched to ECO)

```
1/18/2023, 10:04:34 AM | information | Waiting - EV Connected | 9.25 kWh | 24.1 °C
1/18/2023, 9:12:39 AM | information | Charging | 5.00 kWh | 22.4 °C
1/18/2023, 9:12:36 AM | information | EV Connected | 5.00 kWh | 22.4 °C
1/18/2023, 9:12:32 AM | information | Waiting - EV Connected | 5.00 kWh | 22.4 °C
1/18/2023, 8:11:41 AM | information | Charging | 0.00 kWh | 14.5 °C
1/18/2023, 8:11:38 AM | information | Charging | 0.00 kWh | 14.5 °C
1/18/2023, 8:11:35 AM | information | EV Connected | 0.00 kWh | 14.5 °C
```

KipK, 2023-01-29:
Yeah I got the same issue for a while using limits, never could make it to behave properly.
It always restart few sec after disabling state

jeremypoulter, 2023-01-29:
Limits are still implemented on the EVSE, they are the last main feature that needs to be moved to the ESP32.

KipK, 2023-01-29:
@jeremypoulter I  activated today the limits select box on new UI, using same rapi method as UI v1. 
Should I keep them disabled for now If it was already not working for a while ?

jeremypoulter, 2023-01-29:
May be best

KipK, 2023-02-15:
You  can mark this issue as solved, handled by new Limit engine and UI v2.
Do we need to port it to UI1? I guess it's worthless effort.

closed 2023-03-30 by glynhudson