OpenEVSE/openevse_esp32_firmware #520: featurer request : Enable/Disable Current Shaper via MQTT

FabteslAir, 2023-01-23:
Hi,

When i want to use manual command with current limit at 6A via MQTT and Current Shaper enable, i can't because Current Shaper pass over manual control.

Can you add enable/disable current shapper via MQTT feature, so I could cut the current shaper before doing the manual control with a fixed current.

thanks

jeremypoulter, 2023-01-23:
The current shaper should just be setting the max current allowing other services like MQTT or divert to alter the current between 6A and that max... or at least I thought we had made that change, @KipK ?

KipK, 2023-01-23:
There's a way to temporary disable / enable shaper on Mqtt using the /shaper endpoint
Also shaper is now controlling max_current so you can adjust using charge_current at the same time

FabteslAir, 2023-01-26:
Hi, Thanks,

I use Jeedom to control my Openevse.

the shaper option is very interesting, but during the day I want to limit the load to 6A max, except with shaper active, when I adjust the current via openevse plugins (in Wifi API setting) the intensity is limited, but shaper overrides the current limit a few seconds after the command.

/shaper is only information or I can set it to enable / disable shaper ?

I will try to set max power limitation of shaper ( /shaper_max_pwr) to see if that solves the problem.

Regards

KipK, 2023-01-26:
<base-topic>/shaper/set [0 | 1] : temporary enable (1)/ disable (0) current shaper ( doesn't survive reboot )

FabteslAir, 2023-02-01:
Hi,
I have make a test, 
/shaper/set => OK
/shaper_max_pwr/set [kWh max limit]=> no effect

I can Enable/Disable Current Shaper via /shapper/set, via MQTT but when I change Current shaper disable, the current limit is not supported. it requires resetting the current limit 3 seconds after disabling the current shaper.

Now, my script works perfectly

Thanks

good for you

closed 2023-02-01 by FabteslAir