OpenEVSE/openevse_esp32_firmware #527: Mennekes motor still locked when PowerLost

DylanVautrin, 2023-01-25:
Hello, i have an issue with the PowerLoss behaviour, if the power is cutting down during the charge, the Mennekes locker do not unlock, is there any solution to unlock if we loose the power during a charging session ? 
Thanks
Best regards

DylanVautrin, 2023-01-25:
Hi, i've made some test and i found a way to do it, i think it is not the best : 

[image]

I'm using the OpenEVSE with a OCPP backend and i noticed that when the power is cutting down, the OpenEVSE send a Status Faulted with error code Ground Failure, so i detect this state to unlock the Mennekes motor.
Is there a better way to do it ? 
Thanks

chris1howell, 2023-01-25:
Mennekes Motor is a Forward/Reverse locking setup. It requires power to
lock and unlock, which you do not have when PowerLost. The solution
proposed would work with the addition of a backup battery to power the
controller and Mennekes lock while input power is gone.

DylanVautrin, 2023-01-25:
Ok thanks ! At the end, i should have a battery but for now it work well without (with this modification).
Thanks again !

glynhudson, 2023-03-30:
I've seen some Mennekes lock implementations with a hidden physical cable to allow unlocking in the event of power failure.

closed 2023-03-30 by glynhudson