OpenEVSE/openevse_esp32_firmware #529: Time Manager manual bug spam log

KipK, 2023-01-28:
I felt the EVSE unresponsive when setting to manual mode. 
Could reproduce the issue, just set the time to manual, and reboot OpenEvse Wifi , 
if firmware is compiled with  -D ENABLE_DEBUG_TIME, the log is spammed at high frequency by:

`Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0
Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0
Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0
Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0
Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0
Time manager woke: WakeReason_Scheduled
Local time: 28-Jan-23, 22:27:40
_nextCheckTime = 4716
_setTheTime = 0`

jeremypoulter, 2023-01-29:
Happended to be looking in that area, fairly simple fix, not sure it would have had a massive performance hit though

KipK, 2023-01-29:
That was on UI side, console spam was slowing down the browser, thanks

PR #531 (2023-01-29): No longer endlessly loop twhen switching from NTP -> Manual time
Fixes #529
commit a72eaf11 (2023-01-29): No longer endlessly loop twhen switching from NTP -> Manual time

Fixes #529
closed 2023-01-29 by glynhudson