OpenEVSE/openevse_esp32_firmware #536: Suggestion: Generic Automation Claim Tag

fhteagle, 2023-01-30:
Please add "Automation" to the list of claim tag ClientIDs available in:

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/751d8374ab072df83a920295b6723dc8e904d21e/src/evse_man.h#L28

and in:

https://github.com/KipK/openevse-gui-v2/blob/master/src/lib/vars.js

This would be generic for any third party automation making a claim, such as HomeAssistant.

Not sure what the best priority value should be. 500?

KipK, 2023-01-31:
That would deserve yes. But as most devices will use /override endpoint instead of /claims , the service associated with it will always be Manual

jeremypoulter, 2023-01-31:
What I think I might do is allow a specific client id to be sent along with a priority to the claim endpoint, within some limits, probably will have to be less than the override priority

closed 2026-09-21 by github-actions[bot]