OpenEVSE/openevse_esp32_firmware #540: Pilot current drops to 6A for some max current settings

jdunphy2, 2023-02-05:
Firmware 7.1.3, openevse_wifi_v1 just upgraded to 4.1.7
Some settings (32A, 33A) of the max current pulldown are limiting the maximum current to 6A, while others (31A, 34A) are working as expected.  When the limiting occurrs, the debug window shows {"pilot":6}

KipK, 2023-02-09:
Please update to latest dev builds or UIv2 Builds and tell us if it's solved.

glynhudson, 2023-03-31:
Please could you test the least FW version and post a screenshot / video illustrating the issue?

fhteagle, 2023-04-05:
I can partially confirm this on a unit with 4.1.9 / 8.2.2 firmwares. Right after selection of a charge_current setting, my Tesla Model Y will start to throttle up charging, but after 5-25seconds, the Tesla goes back down to 6 amps actual intake. The Tesala app does not refresh fast enough to show the increased amp limit before it goes back down, but the app is very clearly showing only 6 amps offered to the car.

olivpod, 2023-05-30:
Hello

I have this same trouble with my EVSE:

OpenEVSE | 8.2.3
OpenEVSEWIFI | V4.2.2 on node MCU32s

The charge limit sutck at 6A, if i change the mesured curent on the EVSE page, the curent grow up for 2 ou 3 seconde and immediatly down to 6A.

For "solve" i need to reboot my EVSE, and randomly.. this work and stay stable at 32 A... 

a+

olivpod, 2023-06-01:
video of the trouble..

olivpod, 2023-06-01:
vidéo for this bug
https://youtu.be/7pkXfT_V0p4

olivier

KipK, 2023-06-01:
What openevse firmware build is installed ( openevse, openevse_eu , emonevse, ? )

olivpod, 2023-06-01:
hello

For the wifi card : 
[image]

For the EVSE:
openevse 8.2.3 Pre-release

Olivier

olivpod, 2023-06-01:
it's a standard tethered openevse

olivpod, 2023-06-02:
The file name is openevse.hex ( not Emonevse or openevse_EU )
I'm in europe ( france ) and in setting, i correctly set Level 2 ( not auto, not level 1 )

KipK, 2023-06-02:
Can you put the Europe version on it instead, and retest? As US build has auto-service level + different over current settings.

olivpod, 2023-06-02:
Unfortunately no, I also tried the "openevse_EU" version but my AC driven relay does not work (state never change) with the "openevse_EU" firmware

KipK, 2023-06-02:
I would investigate that first as it should work

Also relay only switch  when a car is plugged in.

olivpod, 2023-06-02:
Yes of course, the relay does not change state when the car is connected and a recharge is required by evse with openevse_EU firmware

KipK, 2023-06-02:
You should find why It doesn't work with eu version. There's no reason but something is wrong in your wiring perhaps, as it also fail somehow differently with us fw too. 

Look at the build flag for eu fw, there's no big difference but the over_current threshold and default to level 2. https://github.com/OpenEVSE/open_evse/blob/master/platformio.ini

Have you checked there's no error displayed on the monitoring section? ( Also with eu fw to see why it doesn't switch )

monfiston, 2023-06-23:
GOOD MORNING
  I also have the same problem with my tesla I now power my esp32 with a 1A usb power supply, it works better, but I still have problems, the current is not stable
have you solved your pb?

olivpod, 2023-07-16:
> hello
> 
> I probalby found my problem for the pilot current ! My wiring is ok, the SW too... but not my EVSE AC/DC converter.
> 
> The +12VDC of my converter is not suffisant in term of current and don't support the new current need of the ESP32 wifi card. +12VDC is not monitored by EVSE and obviously, no error are displayed in monitoring section.
> 
> I have plug a new AC/DC converter just for the ESP 32 and the bug is probalby fix.
> 
> I bench this new parameter for few day and come back for confirm..

hello

For me, the additional ACDC converter doesn't fully solve the trouble.
Sometime without any reason, and only at the beginning of the recharge the current stuck at 6amp.
I only need to reboot the wifi module for "unsticking" the current value.

Olivier

olivpod, 2023-08-26:
Hello, after a new update of my scenario on Jeedom.
I now start the recharge with 8amp value.
Since 3 weeks, I do not need to reboot my evse for my 2 ev.

ramirez22-fr, 2025-02-11:
Hello. 
Yes I know, this is an old post. But do you find a solution ? Cause I have exactly the same problem, with openEvse v8.2.2 and WiFi v5.1.2 ...

admiralroflknife, 2026-02-19:
I understand mostly, I am having an issue with relay switching a lot as well. Could you explain the module more