OpenEVSE/openevse_esp32_firmware #545: v4.1.8 PV Divert Function Sudden CHarging Current Reset

pbix, 2023-02-13:
Using the PV Divert function with OpenEVSE v7.1.3 with Wifi v4.1.8 I am experiencing sudden and unexplained drops in the charging current.  I captured the below debug logs from the UI showing this happening.  The Grid (+I/-E) reference remains constant at -500W but suddenly the current drops from 24A to 8A charging.   The "Max Current" setting is 42A for this charger.

I fear this might be a regression in this new version of firmware since it seems some of the "Fixes" are in this area.

`{"divert_update":0,"grid_ie":-500,"charge_rate":22,"trigger_current":3,"voltage":251.6,"available_current":22.2272814,"smoothed_available_current":15.49601972,"pilot":20,"min_charge_end":166}
{"pilot":22}
{"voltage":251.8999939}
{"amp":20240,"voltage":251.9}
{"session_energy":9690.032222,"total_energy":8983.562,"wh":8983562}
{"amp":20240,"voltage":251.9}
{"amp":21560,"voltage":251.9}
{"session_energy":9693.163333,"total_energy":8983.562,"wh":8983562}
{"amp":22440,"voltage":251.9}
{"amp":22440,"voltage":251.9}
{"session_energy":9696.102778,"total_energy":8983.562,"wh":8983562}
{"amp":22440,"voltage":251.9}
{"amp":22440,"voltage":251.9}
{"session_energy":9699.136389,"total_energy":8983.562,"wh":8983562}
{"amp":22440,"voltage":251.9}
{"claims_version":13}
{"divert_update":0,"grid_ie":-500,"charge_rate":24,"trigger_current":3,"voltage":251.9,"available_current":24.42491465,"smoothed_available_current":17.2817987,"pilot":22,"min_charge_end":158}
{"pilot":24}
{"voltage":251.8000031}
{"amp":22220,"voltage":251.8}
{"session_energy":9702.636667,"total_energy":8983.562,"wh":8983562}
{"amp":22220,"voltage":251.8}
{"amp":8360,"voltage":251.8}
{"session_energy":9704.737222,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.8}
{"amp":5940,"voltage":251.8}
{"session_energy":9705.715278,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.8}
{"amp":5940,"voltage":251.8}
{"session_energy":9706.534722,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.8}
{"claims_version":14}
{"divert_update":0,"grid_ie":-500,"charge_rate":8,"trigger_current":3,"voltage":251.8,"available_current":7.925702939,"smoothed_available_current":16.81399392,"pilot":24,"min_charge_end":150}
{"pilot":8}
{"voltage":251.1000061}
{"amp":5940,"voltage":251.1}
{"session_energy":9707.501944,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.1}
{"amp":5940,"voltage":251.1}
{"session_energy":9708.305556,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.1}
{"amp":5940,"voltage":251.1}
{"session_energy":9709.132778,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.1}
{"amp":5940,"voltage":251.1}
{"session_energy":9709.949722,"total_energy":8983.562,"wh":8983562}
{"amp":5940,"voltage":251.1}
{"divert_update":0,"grid_ie":-500,"charge_rate":8,"trigger_current":3,"voltage":251.1,"available_current":7.93123855,"smoothed_available_current":16.36985615,"pilot":8,"min_charge_end":142}`

> $FP 0 0 Charging~22.22A^42
$OK^20
$FP 0 1 Energy~~~~9.7kWh^45
$OK^20
$GG^24
$OK 22220 251800^1E
$GC^20
$OK 6 80 24 42^1E
$FP 0 0 Charging~~22.22A^42
$OK^20
$GG^24
$OK 8360 251800^23
$GU^36
$OK 34937054 8983562^10
$GC^20
$OK 6 80 24 42^1E
$FP 0 0 Charging~~~8.36A^81
$OK^20
$GG^24
$OK 5940 251800^26
$GC^20
$OK 6 80 24 42^1E
$FP 0 0 Charging~~~5.94A^84
$OK^20
$GG^24
$OK 5940 251800^26
$GU^36
$OK 34940575 8983562^16
$GC^20
$OK 6 80 24 42^1E
$FP 0 0 Charging~~~5.94A^84
$OK^20
$FP 0 1 Lifetime~~~~9MWh^6B
$OK^20

KipK, 2023-02-15:
I see current getting down because your excedent is only around 2Amp ( 500w ), looks to be normal behavior, what's wrong?

KipK, 2023-02-15:
Some tips on setting up +I/-e export 
https://openev.freshdesk.com/support/discussions/topics/6000067405

pbix, 2023-02-15:
When using the PV Divert function GRID_IE is the reference to the charger and represents the amount of power up or down that the charger should change.  What is shown above is that with a request to increase charging by 500W the charger suddenly changes down from 22A to 6A.  I have never seen this behavior in v4.1.3 which I have been using for over a year,  When I changed back to v4.1.3 the problem went away.

pbix, 2023-03-09:
I think this issue is related to 
[https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/463](url)

It seems that when a disconnect occurs the reported value from the MQTT server can drop to zero.

Once the connection is more stable this problem reduces greatly in frequency.

closed 2023-03-09 by pbix

KipK, 2023-03-09:
You should post as retain to not miss an event in case of disconnections.