OpenEVSE/openevse_esp32_firmware #547: Charging won't start without a restart

tux43, 2023-02-17:
Hi,
I occasionally have a problem when charging does not start when requested. If I restart the WiFi Firmware it will then being and operate without issue. What debug mode setting should I turn on and copy/paste to help fault find what is causing this.
Many thanks

KipK, 2023-02-19:
Please post Json result of /claims/target when start doesn't work as expected

joverbee, 2023-02-21:
i seem to have the same issue both with v4.1.7 and 4.1.8. With current shaper on an put to 1333W to compensate for measuring current on only 1 phase to get a total of 4kW max total current for the house.

here is the requested dump
`
{"properties":{"state":"disabled","max_current":4,"auto_release":false},"claims":{"state":65548,"max_current":65548}}
`

the debug terminal show
`{"shaper":1,"shaper_live_pwr":315,"shaper_max_pwr":1333,"shaper_cur":4,"shaper_updated":true}
{"amp":0,"voltage":240,"pilot":6,"wh":38234,"session_energy":0,"total_energy":38.234,"temp":190,"temp_max":195,"temp1":false,"temp2":190,"temp3":false,"temp4":false,"state":254,"status":"disabled","flags":1792,"vehicle":1,"colour":6,"manual_override":1,"freeram":185920,"divertmode":1,"srssi":-45,"time":"2023-02-21T19:31:51Z","offset":"+0100","elapsed":3227}
{"shaper":1,"shaper_live_pwr":315,"shaper_max_pwr":1333,"shaper_cur":4,"shaper_updated":true}
{"amp":0,"voltage":240,"pilot":6,"wh":38234,"session_energy":0,"total_energy":38.234,"temp":190,"temp_max":195,"temp1":false,"temp2":190,"temp3":false,"temp4":false,"state":254,"status":"disabled","flags":1792,"vehicle":1,"colour":6,"manual_override":1,"freeram":187200,"divertmode":1,"srssi":-44,"time":"2023-02-21T19:32:21Z","offset":"+0100","elapsed":3227}
{"shaper":1,"shaper_live_pwr":314,"shaper_max_pwr":1333,"shaper_cur":4,"shaper_updated":true}
{"amp":0,"voltage":240,"pilot":6,"wh":38234,"session_energy":0,"total_energy":38.234,"temp":190,"temp_max":195,"temp1":false,"temp2":190,"temp3":false,"temp4":false,"state":254,"status":"disabled","flags":1792,"vehicle":1,"colour":6,"manual_override":1,"freeram":189060,"divertmode":1,"srssi":-43,"time":"2023-02-21T19:32:51Z","offset":"+0100","elapsed":3227}
`

KipK, 2023-02-21:
@joverbee  Are you on threephase setup ? have you enable threephase settings in Evse config tab ?

joverbee, 2023-02-22:
@KipK I feel dumb but I don’t really see where that evse config tab is. Is it in the proces of compiling the firmware for the evse?
In any case, the firmware and hardware measures only 1 phase current and as long as I know that I could work around is what I thought.

KipK, 2023-02-22:
It's on the config / Evse tab on UI v2 only.
It is still not optimal and probably has bugs has I don't have threephase to test, but worth to try.

KipK, 2023-02-22:
seeing /claims/target you've posted, it is shaper that has disabled the evse. 
Enabling is_threephase in /config should calculate the shaper correctly. 
If you're not on UI2 you can manually http POST to /config {"is_threephase": true}

joverbee, 2023-02-22:
Ok, I did set the  threephase:true, this remains after restart. I moved the max power in the current shaper to the desired 4000W. But the car still doesnt start charging. When I disable the current shaper it starts immediately.

BUT: I had set max current to 6A and the charging shows charging at around 5A. Which looked OK to me. However my car (Renault ZOE) indicated that it would take 28h(!) to charge which explains why last morning my battery was alarmingly low. Looking via the digital net meter shows that total consumption of the house is 400W which can't agree with the car being charged at 5A on 240V 3 phase. 
I then switched the max current to 8A and the indicated charge current on openevse changes slightly higher to 6.16A but now also the consumption follows with a consumption of a bit less than 4kW which could be about right (220*6*3).

What is going on? is there a minimum current below which the car or openevse (unlikely because i assume the contactor is on when openevse says it is charging) refuse to charge? If so, does that also mean the current shaping can't work at low levels like 4kW?

KipK, 2023-02-22:
OpenEVSE can't charge under 6Amp, this is the minimum.

joverbee, 2023-02-22:
Ok, but what does that mean for a current shaper? Even at 10A set current there is going to be the situation that someone is cooking and the charge current needs to be reduced. What happens if it needs to be reduced below the 6A minimum? Does the contactor go on an off several times (which could be fine if not too often)?

I am interested in this situation since in Belgium the govt. has decided for the 'capacity tarif' which means that the nice high current charging that I had before (and worked really well with openevse) will be heavily taxed based on the 15 min maximum power you hit in a full month. Hence my interest in a current shaper that could depending on e.g. cooking or other appliances throttle the charge current of the car to remain within say a 4kW max total power for the house.

Now I am curious where the 6A limit comes from. It can't be the contactor so it must be the electronic reading of the current pickup coil, but I see I am drifting far from the original topic.

KipK, 2023-02-22:
shaper disable evse when <6A, and waits a bit before reenabling to prevent the relay.

joverbee, 2023-02-23:
Ok, thanks a lot for the help. I read a bit about the protocol and understand that lower than 6A is not allowed and has nothing to do with openevse.

Today i tried again and the car wouldnt start charging at 4000W max setting for the shaper. In the terminal i see that the calculated  shaper current is 5 which probably means because this is lower than 6 that the charging can't start.
Then i changed to 5000W setting and the charging starts.

My conclusion is that because the 6A limitation a realistic minimum for 3 phase and 240V is 5000W for the shaper max setting.

I’ll watch it in the coming days, hopefully smooth charging while staying within the 5kW boundaries.
Thanks again!

KipK, 2023-03-01:
something I don't have the answer: is PP specs says it's 6amp for each phase or for phase total.
Current implementation takes the limit for one phase only, but I'm not sure it is correct.

joverbee, 2023-03-02:
Sorry i was travelling for work this week so didnt do any charging.
Current 6 A limit is indeed per phase. For lower power charging it would indeed make more sense to charge single phase. But it should also not take longer than a night to charge if I want to be able to drive the next day, so this 5kW is not that crazy. I even have a special plug to test the single phase charging. For cable losses the 3 phase is clearly better but I am not sure about the inverter losses in the car.

KipK, 2023-03-10:
Can this issue be closed ?

joverbee, 2023-03-10:
As far as I am concerned yes. Charging stayed always under 4kW in practice while cooking with a nominal setting of 5kW.

tux43, 2023-03-11:
I still have a problem where mine will not restart unless it is power cycled. It happens when it is Solar PV Divert mode too. If the solar available drops it stops charging (correctly) and then occasionally when power is available again it stays in waiting for solar mode. The only way around this is with a restart. I have actually wired up a wifi relay (Shelly 1) and I now automate a restart when OpenEVSE refuses to charge. This has got around my problem however it still exists, with me anyway.

KipK, 2023-03-11:
there's a delay to prevent wearing of the relay, have you waited long enough ?

tux43, 2023-03-11:
> there's a delay to prevent wearing of the relay, have you waited long enough ?

3-4 mins.

KipK, 2023-03-11:
so can you answer to the question I've asked 3 weeks ago at the top of your issue ? ;)

tux43, 2023-03-11:
> so can you answer to the question I've asked 3 weeks ago at the top of your issue ? ;)

In Solar PV divert mode it just doesn’t start.

KipK, 2023-03-11:
> @tux43 Please post Json result of http://openevseip/claims/target when start doesn't work as expected

tux43, 2023-03-11:
> that was the question

Hi, Thanks, that is clearer, here it is. 

{"properties":{"state":"active","charge_current":13,"auto_release":false},"claims":{"state":65538,"charge_current":65538}}

KipK, 2023-03-11:
I mean I need this data when it fails to start, is it the case here ?

tux43, 2023-03-11:
> I mean I need this data when it fails to start, is it the case here ?

Yes. That is correct. I can video the screen if you like since I can reproduce this

KipK, 2023-03-11:
Just wanted to be sure.  Can you check the safety data if there's no errors ?

KipK, 2023-03-11:
Divert mode ( id: 65538 )  as set the evse to active according to the target endpoint , seeing that I'd say it's a relay stuck or evse module fault, or safety like gfci and co

tux43, 2023-03-11:
I've completed a screen recording where I simulate the issue. 

https://youtu.be/8N3-kH4xK40 

I always seem to have a count of 2 ground errors even when things are running well. You will see I found it starts running again when I switch it to Normal (Fast) and then back to Eco Mode.

[image]

KipK, 2023-04-04:
Can you check your ground earth wil a multimeter ? 

https://electrouniversity.com/how-to-test-ground-with-multimeter/

tux43, 2023-04-10:
> Can you check your ground earth with a multimeter ?

I've checked and 237V between A-N and A-E so no grounding issues. The count doesn't increase more than 2 as per above. It still operates correctly, except doesn't start charging again. I've worked around this by putting a wifi relay inside the box so if it doesn't start charging it does one reboot and all is good.

KipK, 2023-04-11:
What firmware version your OpenEvse module has ?

tux43, 2023-04-11:
> What firmware version your OpenEvse module has ?

OpenEVSE Firmware: 8.2.1
OpenEVSE WiFi Firmware: 4.1.8

tux43, 2023-04-21:
> What firmware version your OpenEvse module has ?

I've upgraded to the latest version of WiFi with the same issue; although the interface looks better ;-) 

OpenEVSE Firmware: 8.2.1
OpenEVSE WiFi Firmware: 4.2.2

KipK, 2023-05-15:
@tux43, what is your EV ? is it Zoe ar Dacia Spring ?

tux43, 2023-05-20:
> @tux43, what is your EV ? is it Zoe ar Dacia Spring ?

I've done a bit more testing. It works perfectly with the Tesla Model 3 however not the Ioniq 5 (which it is the main charging station for). I have to reboot it when charging does not initatiate and then it is fine.

KipK, 2023-05-20:
I have a Ioniq 5 also and never got problem to start a charge.

tux43, 2023-05-20:
> I have a Ioniq 5 also and never got problem to start a charge.

Do you use 'Self Production' mode to moderate charging rate based on solar output? Or some other method?

[image]

KipK, 2023-05-20:
Nope I don't use solar, but OI use shaper , and during winter it modulates a lot.

But never got a failing charge yet. Do you have latest bms update ?

tux43, 2023-05-21:
> Nope I don't use solar, but OI use shaper , and during winter it modulates a lot.

The issue I am experiancing is when solar suplus is not available for 5 minutes or so. When the surplus becomes available it does not continue to charge as per my video. (Look at 4:44 mark) https://youtu.be/8N3-kH4xK40

NB: I am on the newest firmware now with the same problem, just a nicer interface.

> But never got a failing charge yet. Do you have latest bms update ?

Yes, I assume it gets updated from https://update.hyundai.com/AU/EN/Main and I am on the latest. If this doesn't update the BMS please let me know where I should be grabbing the firmware from.

KipK, 2023-05-21:
BMS can only be updated by Hyundai, it's not part of OTA

KipK, 2023-05-21:
Have you disabled the scheduled charge on your ioniq 5? Or I don't remember if there's a setting to start charge automatically on the car too. Should check that also.

KipK, 2023-05-21:
Also check in openEvse Config/EVSE if the sleep mode is correctly set to Pause and not Disabled

tux43, 2023-06-09:
> Also check in openEvse Config/EVSE if the sleep mode is correctly set to Pause and not Disabled

I've just had a bit of time to do some more testing. Confirming that this setting does not change the behaviour. It will start off ok, then when solar surplus is not available it goes back to 0 Amps. When surplus is available it does not pick up current and continue to charge.

NB: This isn't a big deal for me since I have decided to control the OpenEVSE through the API from Home Assistant rather than using the shaper.