OpenEVSE/openevse_esp32_firmware #552: Why is it charging only at 21A?

SergioRius, 2023-02-26:
I've setup current shaper to 6300W (@28A) that is what my grid supports today/now.
The Max current is setup to 32A, and I have no limit configured on my M3LR.

Those are the sensor values been feed to openevse:
[image]

And this is the status during the charge. It tops at @20A and doesn't move from there, up or down:
[image]

Nothing changes. Asides from that grid import, my panels are producing around 2900w ATM, that are going into my home battery. So I can't understand this limit.
Could be something I've done?

PS: If I set the current shaper to 9000w charge raises to 21.12A and nothing more. (dropping to 20.90 at times)
If I disable current shaper, the pilot value reads 28A (Max configured), but the charge still doesn't raise from 21A.

Thanks for your help.

SergioRius, 2023-02-26:
Although I managed to charge at full speed, after a while it drops to 24A.
The temperature is reading 51ºC, it the Openevse throttling on temperature?

Restarting the charge makes it max the charge rate, but after a little while it drops again.

SergioRius, 2023-02-26:
Ok, it seems the car.
My grid seems to be slightly weak, so voltage drops. As soon as voltage drops below 220 the car throttles.
I've adjusted the PV system to give priority to production over grid (it was configured for dawn) and as import lowered, voltage raised and thus charging rate.

The only thing I still don't understand is why the pilot goes to 6A when I select 32A on Openevse. I tried changing service level from auto to 2 and now I also can't select 31A even after revert to auto. I can only select up to 30A.

closed 2023-02-26 by SergioRius

glynhudson, 2023-03-01:
What car are you charging?

SergioRius, 2023-03-02:
> What car are you charging?

It's a Tesla model 3 Long range.