OpenEVSE/openevse_esp32_firmware #553: Issues with connecting to OCPP server

merlinmag, 2023-02-26:
I setup OpenEVSE device with WiFi module, running firmware  4.1.2. The setup is incomplete - there are only OpenEVSE controller and the WiFi module, even Earth connection is missing. It is done on purpose just to experiment with the software.
I have setup Steve OCPP server https://github.com/steve-community/steve and later on also this server https://github.com/mobilityhouse/ocpp . 

While the OpenEVSE is set up to connect to OCPP server,it is unable to connect. Both of the servers report invalid messages coming from Charge Point, hence preventing the connection. 

Does anyone observed such problem?

jeremypoulter, 2023-02-26:
Are you able to check with the [latest](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/latest) build? That has an updated OCPP library

merlinmag, 2023-02-26:
Which bin shall I take for my hardware, I got lost a bit:
[image]

[openevse_wifi_v1_openevse-gui-v2.bin](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/latest/openevse_wifi_v1_openevse-gui-v2.bin)

or 

[openevse_wifi_v1_openevse_wifi_gui.bin](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/latest/openevse_wifi_v1_openevse_wifi_gui.bin)

merlinmag, 2023-02-26:
Problem solved, most probably the Steve was not correctly installed.

closed 2023-02-26 by merlinmag

jeremypoulter, 2023-02-26:
Good to know, for the record, either of those would do, the difference is the UI.