OpenEVSE/openevse_esp32_firmware #554: New GUI and MQTT

doppiaemme, 2023-02-27:
Hi,

I can't manage to have MQTT connected using the new GUI!
While I have several other devices running fine with the broker, I can't manage to make this work.
Checking the log of the mosquitto broker there is never any attempt from the IP address of openevse.
No Idea how to debug it from this side?
thanks in advance for any clue

[image]

[image]

KipK, 2023-03-01:
I use credentials and works ok. Can you show the full mqtt config screen ?

doppiaemme, 2023-03-01:
There is no much to configure to be honest....
it was working before, the only two things that changed are:
-upgraded the new GUI 
-switched from mosquitto anonymous to mosquito with login/passw (as the Homeassistant's mosquitto addon doesn't allow anonymous anymore). Other MQTT devices are perfectly joining the broker, only openevse gives me an error and no trace of attempred logins on mosquitto's logs

[image]

KipK, 2023-03-01:
check your login is not limited to one session only or use different credentials per device

doppiaemme, 2023-03-01:
downgraded to the old GUI and it worked immediately

2023-03-01 22:53:17: New connection from 192.168.3.71:57195 on port 1883.
2023-03-01 22:53:17: New client connected from 192.168.3.71:57195 as openevse-8dd8 (p2, c0, k60, u'mosquitto').

[image]

KipK, 2023-03-01:
gui only store config data in /config endpoint, problem should not come from there. It has nothing to do on how service works on the fw.
What previous version are you trying ?

KipK, 2023-03-01:
can you export the config from /configuraiton/firmware , or just paste the result of http://openevse/config and /status
when it's not working

doppiaemme, 2023-03-01:
I just reflashed v2 and now it works. no idea what happened but seems like something was corrupted during the first upgrade!
thanks anyhow!

KipK, 2023-03-01:
There's a bug laying around clearing some settings during flash upgrade.
I advice to grab a backup from the new UI, add your password key locally in the json file. And import it after an update until it's solved.

pdhoogh, 2023-03-02:
Yes, I found a reset to factory settings and put everything back in worked for me. I have done it on three occasions.

KipK, 2023-03-02:
you shouldn't need to factory reset, just importing again is sufficient.

closed 2023-03-07 by glynhudson